stupidloud/nanopi-openwrt #1: 感谢

xxyxd, 2020-02-19:
感谢大佬的努力 ，我开了个repo试着编译了一下，R1S-5貌似lean版始终不行
不过原版加passwall没问题了
总之就是感谢啦

klever1988, 2020-02-19:
哈哈 多谢捧场 我也是强迫症 不要客气

xxyxd, 2020-02-21:
不过我看最近的action好多r1s的lean版固件啊，都不知道选哪个了wwwww

klever1988, 2020-02-21:
我设备还没到手无法测试 也不知道哪个版本好用 你试试看吧

xxyxd, 2020-02-21:
> 我设备还没到手无法测试 也不知道哪个版本好用 你试试看吧

诶，那几个lean版区别是啥呢，我准备试试

klever1988, 2020-02-22:
> > 我设备还没到手无法测试 也不知道哪个版本好用 你试试看吧
> 
> 诶，那几个lean版区别是啥呢，我准备试试

相同名字的话，只是追随openwrt的源码更新自动编译，外加小幅度的编译参数修改，比如增删软件包

xxyxd, 2020-02-23:
R1S的几个版本体验下来基本是这样：
1. lienol版各方面体验都不错，除了passwall，passwall和ssrplus同样的节点，ssrplus会好上一些，感觉passwall特点还是在负载均衡。
2. 原版体验参考lienol版
3. lean版单ssrplus体验确实不错，但是资源占用还有pt下载体验比不过lienol版和原版，lean版的flow加速也无法开启。

klever1988, 2020-02-24:
flow没有内核支持是开启不了的，友善是独立内核

使用习惯推荐lean，lienol的好处是温度可以降几度

xxyxd, 2020-02-24:
> flow没有内核支持是开启不了的，友善是独立内核
> 
> 使用习惯推荐lean，lienol的好处是温度可以降几度

好的 , 感谢解答 , 十分感谢

closed 2020-02-24 by xxyxd