stupidloud/nanopi-openwrt #5: 怎么集成双L大的进来

iconycc, 2020-02-23:
有没有方法把Li大的软件特别是L大里没有的一起组合在一起然后加到这官方里编译？目前想L大的基础上加个Passwall，但直接编译L大的没有3328的选，应该是op官方已经不支持了对吧。昨晚刷了Li大的那个包，passwall一开始正常一点，后面再设置就一起连不上外网不知道哪出错了

iconycc, 2020-02-23:
听说IPV6加了有时会出问题，是否考虑出个无IPv6的？

klever1988, 2020-02-24:
两个插件杂交弄不来，分开变异吧

ipv6没事的 Lean给dnsmasq打了补丁 默认屏蔽掉ipv6的域名解析了

lymatthew, 2020-03-04:
Li大有一个分支是在L大固件基础上集成了Passwall的，可以直接编译这个分支：
https://github.com/Lienol/openwrt/tree/dev-lean-lede

klever1988, 2020-03-18:
@iconycc @lymatthew 

https://github.com/klever1988/nanopi-openwrt/releases/tag/R2S-testcombined_with_openclash-2020-03-17

测试吧

lymatthew, 2020-03-18:
> @iconycc @lymatthew
> 
> https://github.com/klever1988/nanopi-openwrt/releases/tag/R2S-testcombined_with_openclash-2020-03-17
> 
> 测试吧

3-16的版本已经用了一天多了，passwall跑地很稳，SSR+原生的应该更没问题。openclash不会配置，不测了。

klever1988, 2020-03-18:
@lymatthew 了解 感谢感谢

fanck0605, 2020-03-25:
> 两个插件杂交弄不来，分开编译吧
> 
> ipv6没事的 Lean给dnsmasq打了补丁 默认屏蔽掉ipv6的域名解析了

怪不得lean的版本没法用ipv6

closed 2020-03-31 by klever1988