stupidloud/nanopi-openwrt #8: 为什么我添加了插件进去就编译失败了

zia123-ai, 2020-02-27:
添加了这些插件
CONFIG_PACKAGE_luci-app-accesscontrol=y
CONFIG_PACKAGE_luci-app-acme=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-arpbind=y
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-baidupcs-web=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-familycloud=y
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-flowoffload=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
CONFIG_PACKAGE_luci-app-kodexplorer=y
CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-mwan3helper=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-app-nlbwmon=y
CONFIG_PACKAGE_luci-app-nps=y
CONFIG_PACKAGE_luci-app-openvpn=y
CONFIG_PACKAGE_luci-app-openvpn-server=y
CONFIG_PACKAGE_luci-app-pptp-server=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-ramfree=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Simple_obfs=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray_plugin=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
CONFIG_PACKAGE_luci-app-syncdial=y
CONFIG_PACKAGE_luci-app-tinyproxy=y
CONFIG_PACKAGE_luci-app-transmission=y
CONFIG_PACKAGE_luci-app-travelmate=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-unblockneteasemusic-mini=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-app-v2ray-server=y
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-vpnbypass=y
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-app-webadmin=y
CONFIG_PACKAGE_luci-app-wireguard=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-app-xlnetacc=y
CONFIG_PACKAGE_luci-app-zerotier=y

klever1988, 2020-02-27:
出错日志来一下

stevenlinfeng, 2020-02-28:
0227 R2S固件，重启会无法启动。需要断电才可以启动

zia123-ai, 2020-02-28:
就显示这个 `Process completed with exit code 1.`
是不是因为添加插件太多了所以编译不成功啊？
我试了好多次了，桌面版ubuntu,谷歌云,DO,都编译失败，很无语厄。
求解决

klever1988, 2020-02-28:
> 0227 R2S固件，重启会无法启动。需要断电才可以启动

我重启过 可以的 你重刷不保存设置试试看

klever1988, 2020-02-28:
> 就显示这个 `Process completed with exit code 1.`
> 是不是因为添加插件太多了所以编译不成功啊？
> 我试了好多次了，桌面版ubuntu,谷歌云,DO,都编译失败，很无语厄。
> 求解决

要有上下文才知道为什么失败的

zia123-ai, 2020-02-28:
不知道为什么我只留下r1s_lean版的yml的actions文件后，提交后actions再也不动了，其他都可以正常
🤣🤣

zia123-ai, 2020-02-28:
[image]
[image]

zia123-ai, 2020-02-28:
就是这个样子提交了，r1s_lean版的那个动都不动了,我试了下其他的yml文件，能正常运行🤣

zia123-ai, 2020-02-28:
我没有添加插件也失败了，然后就变这样了，怎么办。。
[image]

closed 2020-02-28 by zia123-ai