stupidloud/nanopi-openwrt #14: 我添加插件后编译失败

zia123-ai, 2020-02-28:
我添加了这么多插件
CONFIG_PACKAGE_luci-app-accesscontrol=y
CONFIG_PACKAGE_luci-app-acme=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-arpbind=y
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-baidupcs-web=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-familycloud=y
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-flowoffload=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
CONFIG_PACKAGE_luci-app-kodexplorer=y
CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-mwan3helper=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-app-nlbwmon=y
CONFIG_PACKAGE_luci-app-nps=y
CONFIG_PACKAGE_luci-app-openvpn=y
CONFIG_PACKAGE_luci-app-openvpn-server=y
CONFIG_PACKAGE_luci-app-pptp-server=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-ramfree=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Simple_obfs=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray_plugin=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
CONFIG_PACKAGE_luci-app-syncdial=y
CONFIG_PACKAGE_luci-app-tinyproxy=y
CONFIG_PACKAGE_luci-app-transmission=y
CONFIG_PACKAGE_luci-app-travelmate=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-unblockneteasemusic-mini=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-app-v2ray-server=y
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-vpnbypass=y
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-app-webadmin=y
CONFIG_PACKAGE_luci-app-wireguard=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-app-xlnetacc=y
CONFIG_PACKAGE_luci-app-zerotier=y

zia123-ai, 2020-02-28:
你可以帮我看下吗？怎么解决
这是链接 https://github.com/zia123-ai/nanopi-openwrt/runs/474257358?check_suite_focus=true

klever1988, 2020-02-28:
呃 我也看不出来 粗看是runc配置编译参数不通过
你可以通过make menuconfig找到哪个包依赖runc 一直往上找找到依赖他的插件 屏蔽之

zia123-ai, 2020-02-28:
就只显示这个，我找不到哪个插件用上这个runc了,这种情况是不是编译不了了
```
2020-02-28T12:13:40.7971964Z CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80:runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include/fortify -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80:runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include/fortify -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include " LDFLAGS="-L/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/lib -L/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/usr/lib -L/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib -znow -zrelro " GOPATH=/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build GOCACHE=/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/cache GOTMPDIR=/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/tmp GOROOT_FINAL=/usr/lib/go-1.13 CC=aarch64-openwrt-linux-musl-gcc CXX=aarch64-openwrt-linux-musl-g++ 	GOOS=linux GOARCH=arm64 GO386=387 GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CGO_CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80:runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/usr/include -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include/fortify -I/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80:runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" make  -C /home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/runner/work/nanopi-openwrt/n [... cut ...]
2020-02-28T12:13:40.8064302Z make[1]: Leaving directory '/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt'
2020-02-28T12:13:40.8064694Z /home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-h5/friendlywrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-02-28T12:13:40.8069393Z Command exited with non-zero status 1
2020-02-28T12:13:40.8069768Z you take 3:46:56 to build friendlywrt
2020-02-28T12:13:40.8148979Z ##[error]Process completed with exit code 1.
```

TheStoneOasis, 2020-02-29:
> 就只显示这个，我找不到哪个插件用上这个runc了,这种情况是不是编译不了了
>
> ```
> 2020-02-28T12:13:40.7971964Z CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable [...]

我也遇到了一样的情况，我搜了openwrt 编译，runc等关键词，感觉可能是docker依赖runc的问题，正在尝试。

zia123-ai, 2020-02-29:
> 我也遇到了一样的情况，我搜了openwrt 编译，runc等关键词，感觉可能是docker依赖runc的问题，正在尝试。

我之前去掉了docker编译了下，似乎貌似不行额，也可能云编译有点问题

TheStoneOasis, 2020-02-29:
> 我之前去掉了docker编译了下，似乎貌似不行额，也可能云编译有点问题

我把samba4取消后，编译成功了。

zia123-ai, 2020-02-29:
> 我把samba4取消后，编译成功了。

万分感谢，我也去试试

TheStoneOasis, 2020-03-01:
> 呃 我也看不出来 粗看是runc配置编译参数不通过
> 你可以通过make menuconfig找到哪个包依赖runc 一直往上找找到依赖他的插件 屏蔽之

我发现编译好的每个固件里都含有一些在workflow里没添加的ipk，比如sqm，watchcat等等，请问我不想要这些默认ipk的话，在编译时该在哪个文件里删除相关语句呢？

TheStoneOasis, 2020-03-01:
> 万分感谢，我也去试试

我发现编译好的每个固件里都含有一些在workflow里没添加的ipk，比如sqm，watchcat等等，请问我不想要这些默认ipk的话，在编译时该在哪个文件里删除相关语句呢？

zia123-ai, 2020-03-01:
> 我发现编译好的每个固件里都含有一些在workflow里没添加的ipk，比如sqm，watchcat等等，请问我不想要这些默认ipk的话，在编译时该在哪个文件里删除相关语句呢？

你试一试把那个包设置成 `CONFIG_PACKAGE_XXXX is not set`

TheStoneOasis, 2020-03-01:
> 你试一试把那个包设置成 `CONFIG_PACKAGE_XXXX is not set`

就比如watchcat，我的yml文件里都没添加它（没这个语句，自然也没必要设置not set)，但编译成功后依然有这个插件，我查看了各个文件，watchcat存在于ris_linoel.seed文件里，我把它删了，看行不行吧。

zia123-ai, 2020-03-01:
> 就比如watchcat，我的yml文件里都没添加它（没这个语句，自然也没必要设置not set)，但编译成功后依然有这个插件，我查看了各个文件，watchcat存在于ris_linoel.seed文件里，我把它删了，看行不行吧。

还是去问作者把，这个我不会呢🙃

closed 2020-03-03 by zia123-ai