stupidloud/nanopi-openwrt #17: r2s固件，实测无法通过

lsl330, 2020-02-29:


lsl330, 2020-02-29:
测试了r2s的好多固件都有这问题的存在，可以方便排查下原因么？

我看大雕的回复是“和 NET NS 内核选项有关，Docker 会自动选上这个”
不知道是不是因为编译时docker的原因导致这问题？

klever1988, 2020-02-29:
> 测试了r2s的好多固件都有这问题的存在，可以方便排查下原因么？
> 
> 我看大雕的回复是“和 NET NS 内核选项有关，Docker 会自动选上这个”
> 不知道是不是因为编译时docker的原因导致这问题？

[image]
是这样？

lsl330, 2020-02-29:
是的，但我测试智能连接路由的局域网ip（192.168.2.1），nas的就连接不上（192.168.2.3）

lsl330, 2020-02-29:
又测试了一下，Lienol版固件可以连上，lean的固件还是不行

klever1988, 2020-02-29:
> 又测试了一下，Lienol版固件可以连上，lean的固件还是不行

这个 我目前没有交换机 没有测试条件了
你可以尝试在 http://friendlywrt/cgi-bin/luci/admin/network/hosts 添加一条映射试试看

lsl330, 2020-02-29:
> > 又测试了一下，Lienol版固件可以连上，lean的固件还是不行
> 
> 这个 我目前没有交换机 没有测试条件了
> 你可以尝试在 http://friendlywrt/cgi-bin/luci/admin/network/hosts 添加一条映射试试看

lienol版可以连上，lean不行，我看了下config文件，lienol版没有编译docker，lean编译了，不知道这是不是原因

lsl330, 2020-03-04:
刚测试minimal固件也可以局域网端口映射，谢谢编译如此好的固件

closed 2020-03-05 by klever1988