stupidloud/nanopi-openwrt #28: r2s刷lean会出现ipv6地址导致无法上网

NathanDai, 2020-03-03:
ip突然变成ipv6 然后无法上网 无法打开ipv4网站

klever1988, 2020-03-03:
登录ssh执行以下命令禁用ipv6

uci set 'network.lan.ipv6=off'
uci set 'dhcp.lan.dhcpv6=disabled'
/etc/init.d/odhcpd disable
uci commit

NathanDai, 2020-03-03:
执行了命令 然后重启还是不行 使用的是最新的 R2S-Lean-2020-03-03
[image]
[image]

klever1988, 2020-03-03:
[image]
把这个地方的地址也删除保存
[image]

NathanDai, 2020-03-05:
感谢 可以完美解决

closed 2020-03-05 by NathanDai