stupidloud/nanopi-openwrt #42: 怎么添加其他插件到r1s h5的版本里面

zia123-ai, 2020-03-10:


zia123-ai, 2020-03-10:
你更新速度太快了🤣

klever1988, 2020-03-10:
添加编译参数到app_config.seed就行了

zia123-ai, 2020-03-10:
现在r1s flow还是不能用吗

zia123-ai, 2020-03-10:
编译了会不会没用

CWZ7605, 2020-03-10:
> 
> 
> 现在r1s flow还是不能用吗

FLOW一直是正常的。如果你fork过去执行不成功，你可以把release部分更换为action中输出即可。

zia123-ai, 2020-03-11:
> > 现在r1s flow还是不能用吗
> 
> FLOW一直是正常的。如果你fork过去执行不成功，你可以把release部分更换为action中输出即可。

我说的是这个,flow加速在r1s里面用不了
                                                                                                                                                                 [image]

klever1988, 2020-03-11:
r1s目前没有打补丁，我尝试打了补丁了，但系统很不正常，连关机重启都会卡住。暂时就这样吧。因为lan口也就350M的峰值，用不到flow

zia123-ai, 2020-03-11:
> r1s目前没有打补丁，我尝试打了补丁了，但系统很不正常，连关机重启都会卡住。暂时就这样吧。因为lan口也就350M的峰值，用不到flow

好 谢谢大佬😁

closed 2020-03-14 by zia123-ai