stupidloud/nanopi-openwrt #47: 大神怎么直接在路由器里面直接更新固件

zia123-ai, 2020-03-11:
是否有插件可以解决

klever1988, 2020-03-11:
把固件下载到tmp分区
然后gunzip 固件 pipe到dd /dev/mmcblk0

sunguanghui, 2020-03-12:
大神，发现一个问题。
下载的固件包是zip格式，无法直接在openwrt里解压，用opkg去搜索也没有可用的解压软件。
有什么方法可以解决这个问题吗？是否考虑把zip的解压软件打进固件里去？
※gunzip可以解压固件zip里面的.gz压缩包，但解压不了zip

sunguanghui, 2020-03-12:
关键命令：dd if=/mnt/mmcblk0p3/download/lienol-20200312.img of=/dev/mmcblk0

sunguanghui, 2020-03-12:
已测试，可以在线升级

klever1988, 2020-03-12:
gunzip -dc /tmp/upload/rom.img.gz > /dev/mmcblk0
echo 1 > /proc/sys/kernel/sysrq
echo b > /proc/sysrq-trigger

klever1988, 2020-03-12:
@sunguanghui 你把文件放在tf卡上，同时读写，很容易出错的。你应该把压缩包放在/tmp分区，这个分区是内存提供，gunzip输出到卡上就可以了。下面两条命令是强制重启，避免openwrt正常重启时保存设置写入文件到sd卡损坏新刷的固件。

sunguanghui, 2020-03-13:
感谢大神

q943613777, 2020-03-14:
> @sunguanghui 你把文件放在tf卡上，同时读写，很容易出错的。你应该把压缩包放在/tmp分区，这个分区是内存提供，gunzip输出到卡上就可以了。下面两条命令是强制重启，避免openwrt正常重启时保存设置写入文件到sd卡损坏新刷的固件。

大佬问一下，有没保留配置升级的方法。用了上面那个方法，貌似好像是没有保留配置

sunguanghui, 2020-03-14:
> @sunguanghui 你把文件放在tf卡上，同时读写，很容易出错的。你应该把压缩包放在/tmp分区，这个分区是内存提供，gunzip输出到卡上就可以了。下面两条命令是强制重启，避免openwrt正常重启时保存设置写入文件到sd卡损坏新刷的固件。

按照大神的方法测试，固件刷成功了。但还存在以下几个问题：
1、直接把zip包下载到R1S里，没有zip解压工具，必须先在外面解压后，再上传.gz压缩包到R1S里面。
2、固件在线升级后，配置信息无法保留，全部重新配，太折腾了。有没有办法可以保留配置？
3、不同版本互刷后，问题比较多。比如lienol的刷成lean的，经常不稳定，容易掉线等。

大神有没有思路解决以上几个问题？

klever1988, 2020-03-15:
第一个不是问题呐，你就解压在上传gz就好了
保存配置是无解的，r1s/r2s不是op官方支持的设备，只有从底层进行写卡操作进行在线刷机，设置是肯定会丢失的
第三，你刷完一定要立刻执行后两条命令重启，不能有片刻等待，最好的方式是复制三行命令，粘贴到ssh窗口让它连续执行

tianjianping, 2020-03-18:
请教一下，目前能不能恢复出厂设置呢？
使用firstboot命令会有如下提示：
`root@ChongRouter:~# firstboot
This will erase all settings and remove any installed packages. Are you sure? [N/y]
y
MTD partition 'rootfs_data' not found
`
还有没有其它的方法

klever1988, 2020-03-18:
不能 因为这种嵌套userland的运行方式和op原始的运行方式不同 除了重刷没有别的重置办法

当然 你可以反过来做 你可以刷完新固件的时候备份一下初始状态  有毛病的时候拿这个初始备份去还原也差不多

closed 2020-03-31 by klever1988