stupidloud/nanopi-openwrt #48: R2S通过LAN口无法上网

NathanDai, 2020-03-11:
R2S刷入 R2S-Lean-2020-03-11固件会出现无法上国内网络

具体情况：
路由器可以正常连上网络没有问题
[image]

连上路由器的设备无法访问只能打开内网ip无法上国内网络
[image]

打开路由器的科学上网的插件 gwf模式可以打开google却打不开baidu
全局模式 可以正常使用 

期间尝试过的解决方法:
重启，重启后会临时正常一段时间，然后又无法上网

关闭科学上网，无法解决

关闭Turbo ACC 网络加速没有效果

klever1988, 2020-03-11:
在防火墙那边关闭fullcone试试

NathanDai, 2020-03-12:
请问怎么关闭 是在这里吗
[image]

klever1988, 2020-03-12:
[image]

NathanDai, 2020-03-12:
关闭这个好像解决了 我测试两天看看

NathanDai, 2020-03-12:
感谢您的帮助 但是这个方法没法解决 我刷回R2S-Lean-2020-03-12-Minimal 这个固件目前没有任何问题 希望您可以在这个固件上加上docker 
再次感谢您的帮助

lsl330, 2020-03-13:
minimal就是精简的，应该不会加docker之类的软件，要多功能还是下载完整版吧

closed 2020-03-13 by NathanDai

klever1988, 2020-03-13:
把多拨和负载均衡插件卸载掉试试看