stupidloud/nanopi-openwrt #49: 刷入r1s h3固件，内存可用数减少的问题

q943613777, 2020-03-11:
今天将r1s h3的固件刷入r1 h3的机器后发现内存的可用数会一直慢慢减少，不清楚是什么问题导致。
固件：FriendlyWrt_20200311_NanoPi-R1S-H3_armhf_sd.img.gz

q943613777, 2020-03-11:
[image]

klever1988, 2020-03-11:
ssh登陆路由器，输入top -d1，观测一下进程的资源占用，特别是这两个数值，看一下什么进程资源占用大。

[image]

q943613777, 2020-03-12:
> ssh登陆路由器，输入top -d1，观测一下进程的资源占用，特别是这两个数值，看一下什么进程资源占用大。
> 
> [image]

昨晚在固件的“进程”中查看了一下
发现是adbyby的插件内存占用率达到了10。可能是广告过滤大师使用了全局模式造成。后面改成了其他模式，运行了一个晚上内存可用数稳定在80%左右。

closed 2020-03-13 by klever1988