stupidloud/nanopi-openwrt #50: RTL8821CU被识别为存储设备，没有被识别为无线网卡

raiuka, 2020-03-13:
在3.12的minimal的版本中，RTL8821CU被识别为U盘，无法被识别为无线网卡：

Dmesg log如下：

[  196.802565] usb 2-1: new high-speed USB device number 4 using ehci-platform
[  196.831758] usb 2-1: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
[  196.832563] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  196.833234] usb 2-1: Product: DISK
[  196.833567] usb 2-1: Manufacturer: Realtek
[  196.835974] usb-storage 2-1:1.0: USB Mass Storage device detected
[  196.837330] scsi host0: usb-storage 2-1:1.0
[  197.857895] scsi 0:0:0:0: Direct-Access     Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[  197.861311] sd 0:0:0:0: [sda] 16000 512-byte logical blocks: (8.19 MB/7.81 MiB)
[  197.864832] sd 0:0:0:0: [sda] Write Protect is off
[  197.865310] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[  197.868071] sd 0:0:0:0: [sda] No Caching mode page found
[  197.868589] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  197.903505]  sda:
[  197.912999] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  197.961768] FAT-fs (sda): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!
[  197.992524] FAT-fs (sda): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

CWZ7605, 2020-03-13:
在/etc/usb-mode.json里面，添加0bda:1a2b的项目就可以。您自己琢磨一下

raiuka, 2020-03-13:
> usb-mode.json

我将可以识别的固件里的这个文件copy到minimal的固件中无法生效。。。

klever1988, 2020-03-13:
我自己测试确定能用的网卡是这个

https://item.m.jd.com/product/100004852332.html?wxa_abtest=o&ad_od=share&utm_source=androidapp&utm_medium=appshare&utm_campaign=t_335139774&utm_term=CopyURL

raiuka, 2020-03-13:
> 我自己测试确定能用的网卡是这个
> 
> https://item.m.jd.com/product/100004852332.html?wxa_abtest=o&ad_od=share&utm_source=androidapp&utm_medium=appshare&utm_campaign=t_335139774&utm_term=CopyURL

CF-812AC是否支持，如果能支持我就重新买一个。。。不太想继续折腾了

klever1988, 2020-03-13:
[image]
行是行但是连接速率只有54m   意义很小

xxyxd, 2020-03-13:
我用的8811CU，speedtest跑100M没问题，测试距离大概三米左右

noonnn, 2020-03-15:
> [image]
> 行是行但是连接速率只有54m 意义很小

老哥，这是哪一个型号？
是cf-811ac还是cf-812ac或者是别的
812的速度只有54m吗

klever1988, 2020-03-15:
811啊  我给的连接点过去不是811吗

closed 2020-03-17 by klever1988

hjzfree, 2020-08-29:
> > 我自己测试确定能用的网卡是这个
> > 
> > https://item.m.jd.com/product/100004852332.html?wxa_abtest=o&ad_od=share&utm_source=androidapp&utm_medium=appshare&utm_campaign=t_335139774&utm_term=CopyURL
> 
> CF-812AC是否支持，如果能支持我就重新买一个。。。不太想继续折腾了

你的CF-812AC现在支持了吗？