stupidloud/nanopi-openwrt #51: r1s H5 dockerman编译失败

ulpyuxa, 2020-03-13:
大神，这几天我看R1S H5 Lean版本中有编译dockerman，但是编译后的包里面没有docker，这两天我试着手动安装，发现luci的源里面版本发生了变化，
https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/luci/
[image]
已经是072的版本号了，但是我们的源还是使用的086的版本，导致docker无法编译
https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/luci/luci-lib-json_git-20.068.39101-774101e-1_all.ipk

大神看能否解决docker的编译问题

klever1988, 2020-03-13:
是我手动去掉的，
https://github.com/klever1988/nanopi-openwrt/blob/master/.github/workflows/r1s_lean.yml#L53
因为r1s加上docker会依赖runc，最后runc在编译时会出现一个找不到源代码的蜜汁编译错误，我也搞不懂为什么，只能把docker取消掉了。

你看下r1s lean前几天的错误编译日志就知道了

ulpyuxa, 2020-03-16:
看到了，我去掉了那句话就编译不通过，看来和docker无缘了，我记得3/4号，还是3/6号的版本里面是有docker的，估计那时候还没升级版本。
算了，没有就没有吧。

closed 2020-03-20 by klever1988