stupidloud/nanopi-openwrt #52: Bad Gateway 错误

lidatou1991, 2020-03-13:
刷了大佬最新的Lean R1S H5固件。

路由器可以正常启动，红灯闪烁，并且可以看到Friendlywrt的WiFi信号。

Lan口连接电脑，也显示网络已连接。

现在的问题就是，在浏览器地址栏输入http://friendlywrt/后显示
、Bad Gateway
The process did not produce any response、

请问大佬，默认ip是多少呀？如何才可以登录路由器管理界面呢？

sunguanghui, 2020-03-13:
> 刷了大佬最新的Lean R1S H5固件。
> 
> 路由器可以正常启动，红灯闪烁，并且可以看到Friendlywrt的WiFi信号。
> 
> Lan口连接电脑，也显示网络已连接。
> 
> 现在的问题就是，在浏览器地址栏输入 http://friendlywrt/ 后显示
> `Bad Gateway The process did not produce any response`
> 请问大佬，默认ip是多少呀？如何才可以登录路由器管理界面呢？

192.168.2.1 root/password

lidatou1991, 2020-03-13:
> > 刷了大佬最新的 Lean R1S H5 固件。
> > 路由器可以正常启动，红灯闪烁，并且可以看到 Friendlywrt 的 WiFi 信号。
> > Lan 口连接电脑，也显示网络已连接。
> > 现在的问题就是，在浏览器地址栏输入 http://friendlywrt/ 后显示
> > `Bad Gateway The process did not produce any response`
> > 请问大佬，默认 ip 是多少呀？如何才可以登录路由器管理界面呢？
> 
> 192.168.2.1 root/password

用ip还是没办法登陆，一样的显示bad gatewah。不得已换了lionel版本的r1s固件，然后顺利登陆了。
有两个问题想请教：
1、lionel版本的没办法用luci-proto-relay协议么，软件包可以搜索到，安装后在建新接口的下拉菜单中无法选择relayd协议；
2、家里有梅林做主路由，想用r1s 做无线中继，梅林开启wds，r1s没办法设置成client（wds）+ap模式么？总是在第二个ap模式开启后，两个无线都变成master模式。

谢谢大佬解答。

klever1988, 2020-03-13:
r1s这几天都没玩 我等下刷一下试试看

klever1988, 2020-03-13:
[image]
 刚刷 能用 你502应该是luci启动不正常 换个卡重新刷试试
op做wds client我也不知道怎么设置的，建议你老老实实设置成二级路由和ap比较简单
另外 r1s无线和r2s都只能开到54m 用它做ap感觉会慢到吐血 而且r1sr2s的无线网卡驱动都很奇怪 不小心设置错误网卡启动失败 网卡就彻底掉线无法重新救回来 我仓库里有个fix_wifi.sh的脚本 如果网卡下线了你可以执行以下然后重启看看

closed 2020-03-15 by klever1988