stupidloud/nanopi-openwrt #55: 2020.03.14-Minimal不能安装openclash

massivehard2014, 2020-03-15:
无法下载openclash相关的依赖coreutils-nohup、curl
[image]

klever1988, 2020-03-15:
我也不懂为什么清华大学的镜像源为什么会出现各种checksum和size mismatch，你有两个办法，一种是wget 那些ipk包，然后本地opkg install xxx.ipk安装，或者是修改/etc/opkg/distfeeds.conf，把域名改成op官方的downloads.openwrt.org再update一下就可以安装了

massivehard2014, 2020-03-15:
> 我也不懂为什么清华大学的镜像源为什么会出现各种checksum和size mismatch，你有两个办法，一种是wget 那些ipk包，然后本地opkg install xxx.ipk安装，或者是修改/etc/opkg/distfeeds.conf，把域名改成op官方的downloads.openwrt.org再update一下就可以安装了

嗯，直接从网站的packages库里手动下载相关的依赖安装是没有问题的。
还有就是我记得前几天的版本里openclash相关的依赖已经集成了，openclash可以直接安装，现在还要下载安装依赖，不知是怎么回事(之前用的应该是3月初的版本)…

klever1988, 2020-03-15:
减肥给剪掉了，lean的150M肥版估计带了依赖包

closed 2020-03-16 by massivehard2014