stupidloud/nanopi-openwrt #57: 3.15 minimal 版CPU 频率显示错误

raiuka, 2020-03-16:
[image]
见图....

klever1988, 2020-03-16:
这没错啊 而且跟x86显示的格式还是一样的

raiuka, 2020-03-16:
> 这没错啊 而且跟x86显示的格式还是一样的

如果是MHZ应该是1296 MHZ，如果是1296000的话应该是kHZ

klever1988, 2020-03-16:
是啊 ！ 低级错误 ！ 明天修复

raiuka, 2020-03-16:
> 是啊 ！ 低级错误 ！ 明天修复

感谢大佬~，唯一的问题就是无线了。

klever1988, 2020-03-16:
现在无线可以用啊！
 连接速率太低是驱动问题 友善都不解决 无解

raiuka, 2020-03-16:
> 现在无线可以用啊！
> 连接速率太低是驱动问题 友善都不解决 无解

对，就是无线速率的问题，感觉只识别到了802.11A/G,没有支持到N和AC

klever1988, 2020-03-16:
识别是识别到了，也可以调到11ac，问题是不支持频带调节

https://github.com/friendlyarm/rtl8821CU/commit/90e48f268d89425abe85cfca042ada606ed84c0f#diff-154a3697a2c53e6d31153688d2b95311

友善的这个修改，使得网卡驱动不支持工作带宽调节，

我试过取消掉这个修改，但是arm编译器通不过。

这好像和什么浮点数编译器参数有关系，这些底层的东西我不太懂，

closed 2020-03-20 by klever1988