stupidloud/nanopi-openwrt #60: R2S-Lean不稳定

pwrliang, 2020-03-17:
表现为使用一天半左右，无法上网，也无法进入ssh/管理页面，只能拔电源重启。
使用的软件：softether, ssr-plus和防火墙-端口转发.

以下是dmesg日志：

'''
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.22 (runner@fv-az12) (gcc version 6.4.0 (ctng-1.23.0-150g-FA)) #1 SMP PREEMPT Sat Mar 14 10:01:11 UTC 2020
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 16 MiB at 0x000000003f000000
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 253952 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 32 pages/cpu s90280 r8192 d32600 u131072
[    0.000000] pcpu-alloc: s90280 r8192 d32600 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
--
[    3.032848] usb 5-1: Product: USB 10/100/1000 LAN
[    3.033275] usb 5-1: Manufacturer: Realtek
[    3.033642] usb 5-1: SerialNumber: 000000000000
[    3.264376] dwmmc_rockchip ff500000.dwmmc: Successfully tuned phase to 14
[    3.265049] mmc0: new ultra high speed SDR104 SDHC card at address aaaa
[    3.267195] mmcblk0: mmc0:aaaa SC16G 14.8 GiB
[    3.272775]  mmcblk0: p1 p2
[    3.550197] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[    3.866316] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    3.876848] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    4.008179] init: Console is alive
[    4.008588] init: - watchdog -
[    5.752787] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    5.761879] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    5.764745] init: - preinit -
[    5.880187] random: jshn: uninitialized urandom read (4 bytes read)
[    5.902358] random: jshn: uninitialized urandom read (4 bytes read)
[    5.918596] random: jshn: uninitialized urandom read (4 bytes read)
[    8.862600] phy phy-ff450000.syscon:usb2-phy@100.0: charger = USB_DCP_CHARGER
[    8.866878] urandom_read: 2 callbacks suppressed
[    8.866892] random: procd: uninitialized urandom read (4 bytes read)
[    9.065271] mount_root: mounting /dev/root
[    9.092904] mount_root: loading kmods from internal overlay
[    9.273279] kmodloader: loading kernel modules from //etc/modules-boot.d/*
[    9.277328] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
[    9.338034] block: attempting to load /tmp/overlay/upper/etc/config/fstab
[    9.338778] block: unable to load configuration (fstab: Entry not found)
[    9.339406] block: attempting to load /tmp/overlay/etc/config/fstab
[    9.340007] block: unable to load configuration (fstab: Entry not found)
[    9.340626] block: attempting to load /etc/config/fstab
[    9.348537] block: extroot: not configured
[    9.350457] urandom-seed: Seeding with /etc/urandom.seed
[    9.375571] procd: - early -
[    9.375885] procd: - watchdog -
[    9.912687] procd: - watchdog -
[    9.913181] procd: - ubus -
[    9.963416] random: ubusd: uninitialized urandom read (4 bytes read)
[    9.964178] random: ubusd: uninitialized urandom read (4 bytes read)
[    9.964767] random: ubusd: uninitialized urandom read (4 bytes read)
[    9.965858] procd: - init -
[   10.106226] urngd: v1.0.2 started.
[   10.131513] random: crng init done
[   10.131837] random: 1 urandom warning(s) missed due to ratelimiting
[   10.203963] kmodloader: loading kernel modules from /etc/modules.d/*
[   10.222679] NET: Registered protocol family 38
[   10.256758] usbcore: registered new interface driver r8152
[   10.318537] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   10.429190] usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[   10.464156] r8152 5-1:1.0 (unnamed net_device) (uninitialized): DT mac addr 5e:4b:1c:76:b1:3d
[   10.499570] r8152 5-1:1.0 eth1: v1.10.11
[   10.619153] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   10.640491] platform regulatory.0: Direct firmware load for regulatory.db.p7s failed with error -2
[   10.641308] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db.p7s
[   10.648126] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   10.671585] rtl8821CU: loading out-of-tree module taints kernel.[   10.749831] usbcore: registered new interface driver rtl8821cu
[   10.775122] NET: Registered protocol family 15
[   10.790114] tun: Universal TUN/TAP device driver, 1.6
[   10.860955] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   11.068834] Mirror/redirect action on
[   11.110924] u32 classifier
[   11.111185]     input device check on
[   11.111513]     Actions configured
[   11.339853] PPP MPPE Compression module registered
[   11.341254] NET: Registered protocol family 24
[   11.342900] usbcore: registered new interface driver rtl8150
[   11.378492] xt_time: kernel timezone is -0000
[   11.398276] usbcore: registered new interface driver brcmfmac
[   11.411954] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   11.418129] Bridge firewalling registered
[   11.421148] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.574681] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   15.302667] br-lan: port 1(eth1) entered blocking state
[   15.303256] br-lan: port 1(eth1) entered disabled state
[   15.304054] device eth1 entered promiscuous mode
[   15.305739] br-lan: port 1(eth1) entered blocking state
[   15.306432] br-lan: port 1(eth1) entered forwarding state
[   15.327134] r8152 5-1:1.0 eth1: Promiscuous mode enabled
[   15.328501] r8152 5-1:1.0 eth1: carrier on
[   15.629430] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet]
[   15.635964] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   15.636908] ttyS ttyS2: 1 input overrun(s)
[   15.637610] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   15.638916] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   16.318448] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   19.743366] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   19.743985] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   19.744169] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   27.098871] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   32.819553] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   33.433136] br-lan: port 2(tap_vpn) entered blocking state
[   33.433844] br-lan: port 2(tap_vpn) entered disabled state
[   33.435073] device tap_vpn entered promiscuous mode
[   33.435882] br-lan: port 2(tap_vpn) entered blocking state
[   33.436399] br-lan: port 2(tap_vpn) entered forwarding state
[   33.828053] bpfilter: read fail -512
[   43.892764] bpfilter: Loaded bpfilter_umh pid 4095
[   60.153108] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   60.154125] ttyS ttyS2: 3 input overrun(s)
[14084.512341] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[14687.757332] bpfilter: read fail -512
[14687.758674] bpfilter: write fail -32
[14687.760059] bpfilter: write fail -32
[14687.760912] bpfilter: write fail -32
[14687.773273] bpfilter: Loaded bpfilter_umh pid 31857
[14687.773933] bpfilter: read fail -512
[14687.774798] bpfilter: write fail -32
[14687.775703] bpfilter: write fail -32
[14687.810399] bpfilter: Loaded bpfilter_umh pid 31862
'''

Lynamic, 2020-03-18:
很多人遇到同样的问题，而且ssrp还会跑流量，无论是v2ray还是trojan，现在用lienol版本没有出现这些问题

klever1988, 2020-03-18:
如果还在用带vssr版本固件的人，赶紧刷新版的吧

https://github.com/klever1988/nanopi-openwrt/blob/master/CHANGELOG.md#2020-03-10

vssr默认开启的socks服务器会让你的路由器变成公共代理，跑到死机的

pwrliang, 2020-03-18:
@klever1988 您好，我跑的应该是10号以后的release，但是还是有死机的情况，估计不是开了socks代理的问题，我看了下ssr-plus里面的socks5代理是没有勾选的。

```
root@FriendlyWrt:~# uname -a
Linux FriendlyWrt 5.4.22 #1 SMP PREEMPT Sat Mar 14 10:01:11 UTC 2020 aarch64 GNU/Linux
```

klever1988, 2020-03-18:
> @klever1988 您好，我跑的应该是10号以后的release，但是还是有死机的情况，估计不是开了socks代理的问题，我看了下ssr-plus里面的socks5代理是没有勾选的。
> 
> ```
> root@FriendlyWrt:~# uname -a
> Linux FriendlyWrt 5.4.22 #1 SMP PREEMPT Sat Mar 14 10:01:11 UTC 2020 aarch64 GNU/Linux
> ```

可以详细地说一下版本吗

pwrliang, 2020-03-18:
@klever1988 版本为：R2S-Lean-2020-03-14-ROM.zip

klever1988, 2020-03-18:
@pwrliang 在之前的版本有死机现象吗？

pwrliang, 2020-03-18:
@klever1988 之前的lean都会有，使用一天左右就会死机。您那里经过长时间的运行会复现这个问题吗？我用过别人编译的lean固件，自己也编译过，都会出现死机问题。

我不太确定到底是固件本身的问题，还是有安全漏洞被攻击。R2S作为了DMZ，但是ssh我设置了仅允许lan访问。
我用softether的桥接模式作为vpn连回家，然后ssrplus做代理，在防火墙设置开放一些端口给nas的plex和nginx使用，仅此而已。

klever1988, 2020-03-18:
我一半都是刷minimal长时间用，但最长也不超过三天，出现过断流，但是仅仅在我插拔usb无线网卡的时候，不过我和你最大的不同，我是光猫拨号，r2s当二级路由使用。 如果不分固件都会死机，那有没有可能是硬件？

至于肥版的固件，我都是release之后自己刷一下用半小时左右就刷掉了。刚才刷了17号的肥版，看负载没什么问题，我继续跑着观察看看吧。

[image]

ostrichii, 2020-03-21:
我用lean的固件也遇到了，两三天就会出现死机，登录不上也上不了网，重启之后也看不到日志，不知道到底是啥原因导致。怀疑是fullconenat，关了跑一段时间看看

Lynamic, 2020-03-27:
> 我用lean的固件也遇到了，两三天就会出现死机，登录不上也上不了网，重启之后也看不到日志，不知道到底是啥原因导致。怀疑是fullconenat，关了跑一段时间看看

尝试关闭flow加速

closed 2020-03-31 by klever1988