stupidloud/nanopi-openwrt #61: minimal版建议加上kmod-tun等内核模块

king050203, 2020-03-17:
minimal版建议加上kmod-tun等内核模块
mini版使用体验很好，但因有vpn回家需求，程序可以自己装，但内核模块木有办法。
这样就可以后补各种应用，不用从胖版一个个删出来。
谢谢。

klever1988, 2020-03-17:
我给testcombined加上openclash好了，测试效率和minimal差不多的话就用testcombined替换minimal，
目前testcombined是基于Lienol处理好的Lean分支代码编译，自带了ssrp和passwall，

KIMIlin997, 2020-03-23:
> minimal版建议加上kmod-tun等内核模块
> mini版使用体验很好，但因有vpn回家需求，程序可以自己装，但内核模块木有办法。
> 这样就可以后补各种应用，不用从胖版一个个删出来。
> 谢谢。

3-22日，minimal版未见到有kmod-tun安装上

klever1988, 2020-03-23:
> > minimal版建议加上kmod-tun等内核模块
> > mini版使用体验很好，但因有vpn回家需求，程序可以自己装，但内核模块木有办法。
> > 这样就可以后补各种应用，不用从胖版一个个删出来。
> > 谢谢。
> 
> 3-22日，minimal版未见到有kmod-tun安装上

你没下载自带openclash插件的版本么

KIMIlin997, 2020-03-23:
[image]
在R2S-testcombined_with_openclash-2020-03-21-ROM中.config中看到kmod-tun没有编译进去

klever1988, 2020-03-23:
好吧  我一直以为这个是openclash的依赖会被自动选上的 我的错

closed 2020-03-23 by king050203