stupidloud/nanopi-openwrt #66: nanopi r1 h3想通过WiFi连上网，但是失败了

honghongtu, 2020-03-18:
nanopi r1 h3想通过WiFi连上网，但是失败了

klever1988, 2020-03-18:
说得一点都不清楚

一般只要能扫到周围ssid就能接入无线

honghongtu, 2020-03-18:
接进去了但是提示无线未开启或未关联，然后nanopi r1s h3的WiFi都消失了。

klever1988, 2020-03-18:
你得把你的ap无线频宽调到20mhz 
 r1s才能接入

honghongtu, 2020-03-18:
[image]

honghongtu, 2020-03-18:
> 你得把你的ap无线频宽调到20mhz
> r1s才能接入

不是这个原因，上级路由器频宽原本就是20MHz，r1s-h3用的是FriendlyWrt_20200317_NanoPi-R1S-H3_armhf_sd 这个固件，只要一接入扫描到的WiFi，WiFi好像就不工作了。

klever1988, 2020-03-18:
你在系统那边看一下，应该有错误日志，贴过来看看

honghongtu, 2020-03-18:
> 你在系统那边看一下，应该有错误日志，贴过来看看

Wed Mar 18 23:21:02 2020 user.notice sysfixtime: saved 'Wed Mar 18 23:21:02 CST 2020' to /dev/rtc0
Wed Mar 18 23:21:06 2020 user.notice sysfixtime: saved 'Wed Mar 18 23:21:06 CST 2020' to /dev/rtc0
Wed Mar 18 23:21:23 2020 user.notice simple-adblock [2597]: simple-adblock is currently disabled.
Wed Mar 18 23:21:23 2020 user.notice simple-adblock [2597]: Run the following commands before starting service again:
Wed Mar 18 23:21:23 2020 user.notice simple-adblock [2597]: uci set simple-adblock.config.enabled='1'; uci commit;
Wed Mar 18 23:21:25 2020 cron.info crond[7884]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:21:26 2020 cron.info crond[7996]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:07 2020 cron.info crond[8548]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:08 2020 cron.info crond[8659]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:17 2020 kern.info kernel: [  204.777358] device wlan0 left promiscuous mode
Wed Mar 18 23:22:17 2020 kern.info kernel: [  204.782134] br-lan: port 2(wlan0) entered disabled state
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a0:c9:a0:fd:cb:15
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Mar 18 23:22:17 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Wed Mar 18 23:22:18 2020 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Wed Mar 18 23:22:18 2020 user.notice root: mac80211.sh->mac80211_prepare_vif phy=phy0 ifname=wlan0
Wed Mar 18 23:22:18 2020 daemon.notice netifd: radio0 (9117): command failed: No such device (-19)
Wed Mar 18 23:22:18 2020 user.notice mac80211: Failed command: iw phy phy0 interface add wlan0 type managed
Wed Mar 18 23:22:18 2020 kern.info kernel: [  206.491588] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Mar 18 23:22:18 2020 daemon.notice wpa_supplicant[9210]: Successfully initialized wpa_supplicant
Wed Mar 18 23:22:19 2020 daemon.notice netifd: Interface 'wwan' is enabled
Wed Mar 18 23:22:19 2020 cron.info crond[9271]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:20 2020 daemon.notice wpa_supplicant[9218]: wlan0: Trying to associate with 40:a5:ef:12:03:9e (SSID='Netsys_039c' freq=2462 MHz)
Wed Mar 18 23:22:21 2020 kern.warn kernel: [  208.777483] br-lan: received packet on eth1 with own address as source address (addr:ca:1b:c8:85:54:a3, vlan:0)
Wed Mar 18 23:22:22 2020 daemon.notice netifd: Network device 'wlan0' link is up
Wed Mar 18 23:22:22 2020 daemon.notice wpa_supplicant[9218]: wlan0: Associated with 40:a5:ef:12:03:9e
Wed Mar 18 23:22:22 2020 daemon.notice netifd: Interface 'wwan' has link connectivity
Wed Mar 18 23:22:22 2020 daemon.notice netifd: Interface 'wwan' is setting up now
Wed Mar 18 23:22:22 2020 daemon.notice wpa_supplicant[9218]: wlan0: CTRL-EVENT-CONNECTED - Connection to 40:a5:ef:12:03:9e completed [id=0 id_str=]
Wed Mar 18 23:22:22 2020 kern.info kernel: [  210.203462] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Mar 18 23:22:22 2020 daemon.notice wpa_supplicant[9218]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Wed Mar 18 23:22:22 2020 cron.info crond[9678]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:22 2020 daemon.notice netifd: wwan (9700): udhcpc: started, v1.30.0
Wed Mar 18 23:22:22 2020 daemon.notice netifd: wwan (9700): udhcpc: sending discover
Wed Mar 18 23:22:23 2020 cron.info crond[9816]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:25 2020 daemon.err modprobe: xt_multiport is already loaded
Wed Mar 18 23:22:25 2020 daemon.notice netifd: wwan (9700): udhcpc: sending select for 192.168.10.101
Wed Mar 18 23:22:25 2020 daemon.err modprobe: xt_connmark is already loaded
Wed Mar 18 23:22:25 2020 daemon.err modprobe: xt_comment is already loaded
Wed Mar 18 23:22:25 2020 daemon.notice netifd: wwan (9700): udhcpc: lease of 192.168.10.101 obtained, lease time 28800
Wed Mar 18 23:22:25 2020 daemon.err modprobe: xt_length is already loaded
Wed Mar 18 23:22:26 2020 daemon.notice netifd: Interface 'wwan' is now up
Wed Mar 18 23:22:42 2020 user.notice firewall: Reloading firewall due to ifup of wwan (wlan0)
Wed Mar 18 23:22:43 2020 cron.info crond[10659]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:46 2020 cron.info crond[11094]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 23:22:47 2020 cron.info crond[11206]: crond (busybox 1.30.0) started, log level 5

klever1988, 2020-03-18:
看日志没有涉及到调频 我也不知道为什么连不上了。。

uci delete wireless.radio0
uci delete wireless.default_radio0
去ssh执行以下上面这两条命令 然后重启 看看网卡能不能恢复

honghongtu, 2020-03-18:
> 看日志没有涉及到调频 我也不知道为什么连不上了。。
> 
> uci delete wireless.radio0
> uci delete wireless.default_radio0
> 去ssh执行以下上面这两条命令 然后重启 看看网卡能不能恢复

WiFi信号强度依然显示是0%。算了不纠结，我还是老实用有线连接。

closed 2020-03-18 by klever1988

honghongtu, 2020-03-19:
youtube安卓客户端，用ssr plus开v2Ray代理连不上网。

honghongtu, 2020-03-19:
我之前反馈错了，现在发现r1s-h3是可以连上周围WiFi，但是不能放热点出来。

klever1988, 2020-03-19:
我汗 我也觉得我r1s无线客户端模式没什么问题的

ap模式是可以 就是我昨天所说 你不能给ap发射调频 设置调频就会掉网卡出错 必须删掉所有无线设置重来

honghongtu, 2020-03-19:
> 我汗 我也觉得我r1s无线客户端模式没什么问题的
> 
> ap模式是可以 就是我昨天所说 你不能给ap发射调频 设置调频就会掉网卡出错 必须删掉所有无线设置重来

这个问题不研究了，现在有个很明确的问题，ssr plus怎么如果不开全局，开绕过大陆或仅代理GFW，QQ和微信都没有网络。

klever1988, 2020-03-19:
qq微信没网络  那国内网站打得开吗

honghongtu, 2020-03-19:
> qq微信没网络 那国内网站打得开吗

打不开

klever1988, 2020-03-19:
@honghongtu 你去防火墙设置那边把fullcone关闭

honghongtu, 2020-03-19:
> qq微信没网络 那国内网站打得开吗

我把ssr plus关了都没网，太神奇了。

honghongtu, 2020-03-19:
> @honghongtu 你去防火墙设置那边把fullcone关闭

我去试试

honghongtu, 2020-03-19:
> @honghongtu 你去防火墙设置那边把fullcone关闭

防火墙里面，启用FullCone-NAT 这个是没勾选的，是不是开启才没网

honghongtu, 2020-03-19:
> @honghongtu 你去防火墙设置那边把fullcone关闭

有点困了，明天再试一下。

klever1988, 2020-03-19:
不要勾选 勾选对部分网络环境会上不去网

honghongtu, 2020-03-20:
> 不要勾选 勾选对部分网络环境会上不去网

刚才弄了一下，发现开如果不开ssr plus使用正常，开了就只有全局模式有网。开了ssrplus再关掉会没网，重启nanopi r1s -h3之后网络恢复。