stupidloud/nanopi-openwrt #71: R1 h3固件的Turbo ACC 网络加速设置运行不了

q943613777, 2020-03-22:
[image]

我记得大佬前几个固件DNS加速是可以的。但17和21号更新的固件DNS运行不了。

klever1988, 2020-03-22:
r1s是一直不能开加速的 这个设备开不开加速没什么影响啦

q943613777, 2020-03-22:
> r1s是一直不能开加速的 这个设备开不开加速没什么影响啦

好吧。我记得之前更新的几次固件DNS加速是可以开启的。最近发现17号和21号的不行了。
还有我发现21号的固件貌似跟之前前几次更新的固件不一样了。少了一些我用不上的插件。
请问大佬21号更新的r1s h3的固件是做了什么改动吗

klever1988, 2020-03-22:
你帮我比较一下不同日期固件包的.config异同
工具可以考虑用这个
https://www.diffchecker.com/

q943613777, 2020-03-23:
0321-0317  https://www.diffchecker.com/ViGYdowc
0321-0313  https://www.diffchecker.com/m1hfi5Dr
0321-0311  https://www.diffchecker.com/4b3beIIf
0321-0310  https://www.diffchecker.com/LJoc8V0l

q943613777, 2020-03-23:
> 你帮我比较一下不同日期固件包的.config异同
> 工具可以考虑用这个
> https://www.diffchecker.com/

0321-0317 https://www.diffchecker.com/ViGYdowc
0321-0313 https://www.diffchecker.com/m1hfi5Dr
0321-0311 https://www.diffchecker.com/4b3beIIf
0321-0310 https://www.diffchecker.com/LJoc8V0l

klever1988, 2020-03-23:
我看了一下21号flowoffload插件dnscache的基础包pdnsd有编译到了
另外bbr应该能用啊，内核是支持的，你进ssh用lsmod | grep bbr我看看

klever1988, 2020-03-23:
试试把卡格式化一下（普通不要快速），然后重新刷一遍试试

q943613777, 2020-03-23:
> 试试把卡格式化一下（普通不要快速），然后重新刷一遍试试

每次更新固件之后会把卡给格式化，所以这个操作是有做到的

q943613777, 2020-03-23:
> 我看了一下21号flowoffload插件dnscache的基础包pdnsd有编译到了
> 另外bbr应该能用啊，内核是支持的，你进ssh用lsmod | grep bbr我看看

这个要晚上回去看

q943613777, 2020-03-23:
> 我看了一下21号flowoffload插件dnscache的基础包pdnsd有编译到了
> 另外bbr应该能用啊，内核是支持的，你进ssh用lsmod | grep bbr我看看

貌似我用了这个指令没反应

klever1988, 2020-03-23:
那lsmod呢

q943613777, 2020-03-24:
> 那lsmod呢

BusyBox v1.30.0 () built-in shell (ash)

  ___    _             _ _    __      _____ _____ 
 | __| _(_)___ _ _  __| | |_  \ \    / / _ \_   _|
 | _| '_| / -_) ' \/ _` | | || \ \/\/ /|   / | |  
 |_||_| |_\___|_||_\__,_|_|\_, |\_/\_/ |_|_\ |_|  
                           |__/                  
 -----------------------------------------------------
 FriendlyWRT SNAPSHOT, r0+2356-c763bf85d9
 -----------------------------------------------------

DYC Build

root@FriendlyWrt:~# lsmod
8189es               1024000  0 
8821cu               1683456  0 
ah4                    16384  0 
ah6                    16384  0 
authenc                16384  0 
brcmfmac              180224  0 
brcmutil               16384  1 brcmfmac
bridge                 94208  0 
cdc_ether              16384  0 
deflate                16384  0 
echainiv               16384  0 
esp4                   20480  0 
esp6                   20480  0 
ip_set                 32768 16 xt_set,ip_set_list_set,ip_set_hash_netportnet,ip_set_hash_netport,ip_set_hash_netnet,ip_set_hash_netiface,ip_set_hash_net,ip_set_hash_mac,ip_set_hash_ipportnet,ip_set_hash_ipportip,ip_set_hash_ipport,ip_set_hash_ipmark,ip_set_hash_ip,ip_set_bitmap_port,ip_set_bitmap_ipmac,ip_set_bitmap_ip
ip_set_bitmap_ip       20480  0 
ip_set_bitmap_ipmac    20480  0 
ip_set_bitmap_port     20480  0 
ip_set_hash_ip         32768  1 
ip_set_hash_ipmark     32768  2 
ip_set_hash_ipport     32768  0 
ip_set_hash_ipportip   32768  0 
ip_set_hash_ipportnet   36864  0 
ip_set_hash_mac        20480  0 
ip_set_hash_net        32768  9 
ip_set_hash_netiface   36864  0 
ip_set_hash_netnet     36864  0 
ip_set_hash_netport    36864  0 
ip_set_hash_netportnet   36864  0 
ip_set_list_set        20480  2 
ip_tables              24576  5 iptable_nat,iptable_raw,iptable_mangle,iptable_filter
ip_tunnel              24576  1 sit
ip6_tables             24576 30 ip6table_nat,ip6table_mangle,ip6table_filter
ip6t_MASQUERADE        16384  0 
ip6t_NPT               16384  0 
ip6t_REJECT            16384  2 
ip6table_filter        16384  1 
ip6table_mangle        16384  1 
ip6table_nat           16384  0 
ipcomp                 16384  0 
ipcomp6                16384  0 
ipt_ECN                16384  0 
ipt_MASQUERADE         16384  2 
ipt_REJECT             16384  2 
ipt_ah                 16384  0 
iptable_filter         16384  1 
iptable_mangle         16384  1 
iptable_nat            16384  1 
iptable_raw            16384  1 
llc                    16384  2 bridge,stp
nf_conntrack           94208 36 nf_nat_pptp,nf_conntrack_pptp,ipt_MASQUERADE,xt_state,xt_nat,xt_helper,xt_conntrack,xt_connmark,xt_connlimit,xt_connbytes,xt_REDIRECT,xt_CT,nf_nat_tftp,nf_nat_snmp_basic,nf_nat_sip,nf_nat_masquerade_ipv4,nf_nat_irc,nf_conntrack_ipv4,nf_nat_ipv4,nf_nat_h323,nf_nat_ftp,nf_nat_amanda,nf_conntrack_tftp,nf_conntrack_snmp,nf_conntrack_sip,nf_conntrack_proto_gre,nf_conntrack_netlink,nf_conntrack_irc,nf_conntrack_h323,nf_conntrack_ftp,nf_conntrack_broadcast,nf_conntrack_amanda,nf_conntrack_ipv6,nf_nat_ipv6,nf_nat_masquerade_ipv6,nf_nat
nf_conntrack_amanda    16384  2 nf_nat_amanda
nf_conntrack_broadcast   16384  1 nf_conntrack_snmpnf_conntrack_ftp       16384  2 nf_nat_ftp
nf_conntrack_h323      45056  3 nf_nat_h323
nf_conntrack_ipv4      16384 43 
[... 66 lines cut ...]
xt_comment             16384213 
xt_connbytes           16384  0 
xt_connlimit           16384  0 
xt_connmark            16384  4 
xt_conntrack           16384 26 
xt_dscp                16384  0 
xt_ecn                 16384  0 
xt_esp                 16384  0 
xt_helper              16384  0 
xt_hl                  16384  0 
xt_length              16384  0 
xt_limit               16384 20 
xt_mac                 16384  0 
xt_mark                16384 38 
xt_multiport           16384  6 
xt_nat                 16384  1 
xt_policy              16384  4 
xt_recent              20480  0 
xt_set                 20480 27 
xt_state               16384  0 
xt_statistic           16384  0 
xt_tcpmss              16384  0 
xt_tcpudp              16384 42 
xt_time                16384  0 
root@FriendlyWrt:~# lsmod

klever1988, 2020-03-24:
modprobe tcp_bbr
试试看

klever1988, 2020-03-24:
试一下今天新编译的版本能开不

q943613777, 2020-03-24:
> 试一下今天新编译的版本能开不

好，晚上回去试

q943613777, 2020-03-24:
> 试一下今天新编译的版本能开不

今天编译的版本，DNS加速又可以运行起来了
[image]

q943613777, 2020-03-24:
> 试一下今天新编译的版本能开不

今天编译的版本试了modprobe tcp_bbr指令，貌似没啥反应
[image]

klever1988, 2020-03-25:
[image]
这两项建议关闭
bbr已经加入内核支持
下载最新版试试看

q943613777, 2020-03-25:
> [image]
> 这两项建议关闭
> bbr已经加入内核支持
> 下载最新版试试看

刷了今天更新的R1s h3的固件，可以运行DNS加速和BBR 加速了
[image]

q943613777, 2020-03-25:
> [image]
> 这两项建议关闭
> bbr已经加入内核支持
> 下载最新版试试看

大神，发现个问题
开启bbr之后，貌似会影响我二级路由器的wifi功能。具体表现就是移动设备上不了网。关闭之后就恢复正常。

klever1988, 2020-03-25:
bbr对wifi确实是有负优化的 网上有讨论

q943613777, 2020-03-26:
> bbr对wifi确实是有负优化的 网上有讨论

好吧。那就是无解了

closed 2020-03-26 by klever1988