stupidloud/nanopi-openwrt #72: SQM 没有cake队列

ostrichii, 2020-03-22:
[image]
之前一直使用SQM中的Cake队列来做QoS，不然挂PT的时候会导致网络拥塞，但是我看现在编译的版本里面没有这个队列选项了
https://www.bufferbloat.net/projects/codel/wiki/Cake/

klever1988, 2020-03-22:
[image]
我看了下21号lean版的.config，是包含了cake的kmod模块的，不知道为什么不可用了

klever1988, 2020-03-26:
cake要内核支持才行，我从未在友善的内核添加cake模块，应该是一直就没有cake队列可用吧？

ostrichii, 2020-03-26:
> cake要内核支持才行，我从未在友善的内核添加cake模块，应该是一直就没有cake队列可用吧？

我记得之前是有的，印象还比较深刻，开了200mCPU使用率在50%+了，如果大佬一直没加那估计是5.12的内核时候了

klever1988, 2020-03-26:
cake属于第三方模块，友善内核不可能预置这玩意儿吧

ostrichii, 2020-03-26:
翻了一下机器，是是我用的大雕的插件和友善的源码编译出来的，不是大佬的固件

klever1988, 2020-03-26:
手动斜眼

你快研究一下怎么增加cake支持，然后push过来

closed 2020-03-31 by klever1988