stupidloud/nanopi-openwrt #75: 一个小的显示问题

raiuka, 2020-03-25:
[image]

删除按钮错误。。

PS：另外关于无线54M的问题，我看到实际上连上去以后使用的协议是802.11a，官方源码好像没这个问题？我有点记不清楚了。所以8821cu在现在的驱动下没有识别到802.11ac。如果强制改为legacy N的话，速度能提升到150M

klever1988, 2020-03-25:
你试过可以150m?

klever1988, 2020-03-25:
[image]
分享一下你的设置，我这边情况是，只要调到N模式，保存应用之后网卡就会彻底掉线无法再驱动起来。。

另外那个按钮错误是主题css问题，你得跟老竭力反馈一下
https://github.com/jerrykuku/luci-theme-argon

raiuka, 2020-03-25:
> [image]
> 分享一下你是怎么设置吧。。
> 
> 我这边情况是，只要调到N模式，保存应用之后网卡就会彻底掉线无法再驱动起来。。
> 
> 另外那个按钮错误是主题css问题，你得跟老竭力反馈一下
> https://github.com/jerrykuku/luci-theme-argon

对，没错，网卡会彻底掉线找不到，这个时候你重启一下R2S，重新上线后就正常了

raiuka, 2020-03-25:
[image]

我设置为legacy N，强制40Mhz频宽

[image]
开启MU-MIMO（虽然单天线没什么卵用），开启Mu-MIMO的时候网卡也会掉，重启就好。

klever1988, 2020-03-25:
还真可以。。。
原来驱动的bug是不支持动态调节频宽，重启调节是可以的。。

raiuka, 2020-03-25:
> 还真可以。。。
> 原来驱动的bug是不支持动态调节频宽，重启调节是可以的。。

剩下就是解决802.11ac协议的问题了。。。802.11a这破协议也实在太老了

closed 2020-03-31 by klever1988