stupidloud/nanopi-openwrt #77: ipv6 not work

iamid0, 2020-03-26:
网络环境
电信网关  --> WiFi路由器(采用ipv6 pass through)--> iPad, ipv4 & ipv6，此时iPad得到的ipv6地址是电信的。访问网络ipv4, ipv6均工作。 

电信网关--> R2S --> WiFi路由器 --> iPad, ipv4 & ipv6，此时iPad得到ipv4正常，ipv6为fd17:d0e5:7ec1开头，iPad无法访问ipv6网络。

检查发现，fd17:d0e5:7ec1开头的地址是。

固件来自R2S-Lienol-2020-03-25。

所有网络接口采用默认配置。 

其中WAN6，可以获取到正常的电信ipv6地址，
LAN口获取的ipv6地址fd17:d0e5:7ec1::1/60，也能给后面的设备下发ipv6地址，但是疑似ipv6 NAT不可用，导致客户端ipv6虽然有地址，但是不工作。

之前使用R1S的Lienol固件中，也是同样的问题。
未测试Lean固件。

klever1988, 2020-03-26:
做二级路由的时候，你需要对openwrt的ipv6设置中继。
你遇到的问题openwrt的使用，和设备是无关的。

iamid0, 2020-03-26:
你好，你说的是在R2S的LAN接口， DHCP服务器 --> IPv6设置中，路由通告设定为 终极模式；DHCPv6服务器设定为 终极模式吗？ 

另， ssh 登录R2S, ifconfig可看到有电信的ipv6地址，测试ipv6连接正常。

klever1988, 2020-03-26:
http://blog.kompaz.win/2017/02/22/OpenWRT%20IPv6%20%E9%85%8D%E7%BD%AE/

用第一种方式进行配置

iamid0, 2020-03-26:
多谢指点。我比较熟悉ipv4 NAT，结合你的资料，我应该能搞定。再次感谢。

klever1988, 2020-03-26:
不客气

ipv6地址足够多的  不要用nat 中继是最好的

iamid0, 2020-03-26:
参照你推荐的连接，采用了ipv6中继模式，客户端可以得到原生ipv6地址，但是客户端访问ipv6网络有很懂问题。 
1 iPhone iPad 可以获取原生ipv6地址，不过访问ip6.me失败，其他ipv4网络正常。
2 Mac 连接同一个WiFi, 可以获取原生ipv6地址，可以ping6到目标主机，ssh登录后，操作不到几秒，然后就断掉了。

我可能没有功夫debug这个问了，直接换用两个WiFi来处理就好了。

再次谢谢你。^_^

closed 2020-03-31 by klever1988