stupidloud/nanopi-openwrt #81: R2S lean固件在“状态”->“概况”页面下CPU一直占用率20%以上，其他页面没有问题

liuenrong, 2020-03-26:
[image]

klever1988, 2020-03-26:
这个很正常，因为这个页面是一个是脚本一直在输出动态数据，需要占用很大的cpu时间

liuenrong, 2020-03-27:
> 这个很正常，因为这个页面是一个是脚本一直在输出动态数据，需要占用很大的cpu时间

谢谢回复，就是这个CPU占用有点太高

klever1988, 2020-03-27:
https://github.com/coolsnowwolf/lede/commit/b2ca31e6b0de42d3b009044edab69dbc7e111967

大雕今天对首页数据生成逻辑做了修改，静待今天的版本看看有没有降低

liuenrong, 2020-03-27:
> https://github.com/coolsnowwolf/lede/commit/b2ca31e6b0de42d3b009044edab69dbc7e111967
> 
> 
> 
> 大雕今天对首页数据生成逻辑做了修改，静待今天的版本看看有没有降低

OK

lernb, 2020-03-27:
> [coolsnowwolf/lede@b2ca31e](https://github.com/coolsnowwolf/lede/commit/b2ca31e6b0de42d3b009044edab69dbc7e111967)
> 
> 大雕今天对首页数据生成逻辑做了修改，静待今天的版本看看有没有降低

之前刚看到今天自动发布的版本，现在看没了，还没下载

lernb, 2020-03-28:
> https://github.com/coolsnowwolf/lede/commit/b2ca31e6b0de42d3b009044edab69dbc7e111967
> 
> 
> 
> 大雕今天对首页数据生成逻辑做了修改，静待今天的版本看看有没有降低

试了27号版的，概况页面CPU占用情况，最低的时候比之前低，最高的比之前高，netdata监控显示总体还是在20%以上

klever1988, 2020-03-28:
那木有办法 习惯就好

klever1988, 2020-03-28:
[image]
我觉得还好啊 minimal一直是3-5%的占用

lernb, 2020-03-28:
> [image]
> 
> 我觉得还好啊 minimal一直是3-5%的占用

Lean版“概况”页面今天的版本是百分之几、百分之三十几的跳，netdata上看就是一直在百分之二十几的浮动

klever1988, 2020-03-28:
进ssh用top命令看看

lernb, 2020-03-29:
> 进ssh用top命令看看

[image]
已经刷了28号版的lean固件，情况好很多了，现在一直在百分之十以下，netdata上看是百分之十左右。不过一开始占用都是在二十以上，ssh进去top看是图中的一直占用25%左右

closed 2020-03-31 by klever1988