stupidloud/nanopi-openwrt #83: 更新脚本BUG

king050203, 2020-03-27:
昨天的固件已下载，准备解压
Archive:  R2S-Minimal-2020-03-26-ROM.zip
   creating: artifact/
  inflating: artifact/.config
  inflating: artifact/FriendlyWrt_20200326_NanoPi-R2S_arm64_sd.img.gz
92.7MiB 0:02:04 [ 759kiB/s] [================================>] 100%
准备解压镜像文件
解压已完成，准备编辑镜像文件，写入备份信息
Saving config files...
备份文件已经写入，移除挂载
准备重新打包
/mnt/mmcblk0p2/FriendlyWrt.img :  9.22%   (999999488 => 92190448 bytes, /tmp/FriendlyWrtupdate.img.zst)
打包完毕，准备刷机
umount: can't unmount /: Resource busy
87.9MiB 0:03:05 [ 485kiB/s] [================================>] 100%
1953124+0 records in
1953124+0 records out
刷机完毕，正在重启...

umount失败不检测，脚本继续强制刷机，导致重启起不来。

klever1988, 2020-03-27:
你的运行版本是lienol版吧，我也遇到不能卸载根分区的情况，不过有时候运气好还能刷进去
总之这种刷法只能做偷懒用，最靠谱还是拔卡刷

king050203, 2020-03-27:
是R2S-Minimal-2020-03-23，还是把卡刷靠谱。

closed 2020-03-27 by klever1988