stupidloud/nanopi-openwrt #85: 多级路由科学上网问题

zia123-ai, 2020-03-27:
大神好,我用2级路由R1S连接1级路由光猫的lan口，再用3级路由连接R1S的LAN口，然后SSR PLUS选择G F W列表模式，结果上不了网，这个问题困扰了好久，但是选择绕过大陆IP模式，3级路由所连接的设备就可以正常上网。

klever1988, 2020-03-27:
你设备是不是设置了自定义dns了

zia123-ai, 2020-03-27:
没了用得都是自动分配的路由DNS

klever1988, 2020-03-27:
最右边那个路由用lan口连接r1s的lan口，关闭路由器的dhcp功能
这样才能让你的r1s catch到客户端的dns解析请求

zia123-ai, 2020-03-27:
> 最右边那个路由用lan口连接r1s的lan口，关闭路由器的dhcp功能
> 这样才能让你的r1s catch到客户端的dns解析请求

这样是把R1S当作旁路由吗

klever1988, 2020-03-27:
> 这样是把R1S当作旁路由吗

这样是把你的无线路由做纯ap使用

zia123-ai, 2020-03-27:
> > 这样是把R1S当作旁路由吗
> 
> 这样是把你的无线路由做纯ap使用

我试了一下 lan口连lan口 结果显示no links，内网DHCP也全关了。。

klever1988, 2020-03-27:
> 我试了一下 lan口连lan口 结果显示no links，路由器的内网DHCP也全关了。。
什么显示no links?显示在哪里？

zia123-ai, 2020-03-27:
> > 我试了一下 lan口连lan口 结果显示no links，路由器的内网DHCP也全关了。。
> 
> 什么显示no links?显示在哪里？

解决了解决了，就是开始那个做纯AP的路由的WAN口显示no links，后来改了Lan口为R1S的网段IP后就可以了

klever1988, 2020-03-27:
也能上网了吧？

zia123-ai, 2020-03-27:
> 也能上网了吧？

能了，我原来以为只能把R1S当旁路有来处理的，谢谢😁

closed 2020-03-27 by klever1988