stupidloud/nanopi-openwrt #91: 大神，刷了26日和27日的版本，FULLCONE NAT加速未运行

monsterdior, 2020-03-27:
麻烦大神看看，26日和27日的版本都没有运行HWNAT，

klever1988, 2020-03-27:
http://friendlywrt/cgi-bin/luci/admin/network/firewall

防火墙那边开启。。

monsterdior, 2020-03-28:
谢谢！。。。折腾一晚上。。。。。大神还有一个问题。装zerotier，发现缺一个基础包，可是在官方源里面找不到了。
Installing zerotier (1.4.6-2) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/zerotier_1.4.6-2_aarch64_cortex-a53.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for zerotier:
 * 	libstdcpp6
 * opkg_install_cmd: Cannot install package zerotier.

klever1988, 2020-03-28:
大包不是有zerotier了吗

monsterdior, 2020-03-28:
用的minimal版本。T_TT_T

klever1988, 2020-03-28:
opkg intall --nodeps zerotier

klever1988, 2020-03-28:
这样试试

monsterdior, 2020-03-28:
能够安装，安装后配置config，之后运行 貌似报了很多错，是不是需要重启才能用？
root@FriendlyWrt:~# /etc/init.d/zerotier start
Generate secret - please wait...
Error loading shared library libminiupnpc.so.17: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error loading shared library libnatpmp.so.1: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error relocating /usr/bin/zerotier-idtool: UPNP_GetSpecificPortMappingEntry: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_AddPortMapping: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetValidIGD: symbol not found
Error relocating /usr/bin/zerotier-idtool: initnatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: freeUPNPDevlist: symbol not found
Error relocating /usr/bin/zerotier-idtool: closenatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetExternalIPAddress: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendpublicaddressrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: getnatpmprequesttimeout: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendnewportmappingrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: upnpDiscoverAll: symbol not found
Error relocating /usr/bin/zerotier-idtool: readnatpmpresponseorretry: symbol not found

klever1988, 2020-03-28:
opkg install zerotier --force-depends

monsterdior, 2020-03-28:
安装的时候还是报错，但是启动没问题了~谢谢大神，我在琢磨一下。感谢
Collected errors:
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:3: Ignoring invalid line: `s#rc/gz openwrt_luci http://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/mipsel_24kc/luci'
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:5: Ignoring invalid line: `s#rc/gz openwrt_packages http://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/mipsel_24kc/packages'
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:6: Ignoring invalid line: `src#/gz openwrt_routing http://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/mipsel_24kc/routing'
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:7: Ignoring invalid line: `src/#gz openwrt_telephony http://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/mipsel_24kc/telephony'
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:8: Ignoring invalid line: `src/g#z omega2_core http://repo.onioniot.com/omega2/packages/core'
 * opkg_conf_parse_file: /etc/opkg/customfeeds.conf:9: Ignoring invalid line: `src/gz# omega2_base http://repo.onioniot.com/omega2/packages/base'

klever1988, 2020-03-28:
你自己改了customfeeds.conf，把它还原就不会报错了

monsterdior, 2020-03-28:
嗯~嗯~谢谢大佬，

forestslin, 2020-03-28:
root@FriendlyWrt:~# /etc/init.d/zerotier start
Generate secret - please wait...
Error loading shared library libminiupnpc.so.17: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error loading shared library libnatpmp.so.1: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error relocating /usr/bin/zerotier-idtool: UPNP_GetSpecificPortMappingEntry: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_AddPortMapping: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetValidIGD: symbol not found
Error relocating /usr/bin/zerotier-idtool: initnatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: freeUPNPDevlist: symbol not found
Error relocating /usr/bin/zerotier-idtool: closenatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetExternalIPAddress: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendpublicaddressrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: getnatpmprequesttimeout: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendnewportmappingrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: upnpDiscoverAll: symbol not found
Error relocating /usr/bin/zerotier-idtool: readnatpmpresponseorretry: symbol not found
  我的也是报错，大神能否告知怎么回事 minimal版

klever1988, 2020-03-28:
> root@FriendlyWrt:~# /etc/init.d/zerotier start
> Generate secret - please wait...
> Error loading shared library libminiupnpc.so.17: No such file or directory (needed by /usr/bin/zerotier-idtool)
> Error [...]

opkg install zerotier --force-depends

forestslin, 2020-03-28:
直接使用这个命令安装就可以了吗？
opkg install zerotier --force-depends

forestslin, 2020-03-28:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for zerotier:
 *      libstdcpp6
 * opkg_install_pkg: Package size mismatch: libminiupnpc is 23992 bytes, expecting 23998 bytes
 * opkg_install_cmd: Cannot install package zerotier.

forestslin, 2020-03-28:
直接使用  opkg install zerotier --force-depends
然后显示缺少依赖，无法安装

klever1988, 2020-03-28:
libstdcpp6这个错误无视它

Package size mismatch遇到这种错误，手动wget ipk包，然后opkg install 本地ipk包方式进行安装

monsterdior, 2020-03-28:
http://koolshare.cn/thread-152561-1-1.html
我安装之后没有web界面~建议参照上面的要求做

forestslin, 2020-03-28:
谢谢，搞定了

closed 2020-03-31 by klever1988