stupidloud/nanopi-openwrt #94: 请问使用Lienol版固件的R2S用户，你们会出现不明原因的死机情况吗？

klever1988, 2020-03-28:
R1S用户可以无视此贴
目前的Lean版以及Minimal版本固件经常收到不明原因的死机报告
具体情况就是状态灯正常，但是web后台连不上，并且“本地连接”会失去ipv4网络；大约在运行固件两天以上才会碰到；如果你有ipv6网络或者ttl，仍然可以连接到后台或者ssh，并且可以执行ping wan口 ip得到operation not permitted的错误信息
目前只能初步猜测是Lean大雕对op实施的优化或者某个插件与friendlylinux内核不兼容致使iptables崩溃

如果Lienol版固件没有这个情况，将考虑采用Lienol1907分支的op合并Lean的ssrp插件进行编译Minimal固件

fanck0605, 2020-03-29:
有死机现象，其实还是 firewall 的问题，ipv6进管理界面，重启firewall就解决

使用环境：r2s 拨号，wan lan 对调，开启了full cone nat，bbr，未启用 flow 加速

复现方式：再接口中点击wan的**连接**按钮，触发重新连接PPPoE，等待 1-2S IPv4 就会崩溃
ps：我是有公网 IP 的概率性获得，所以要手动重新拨号，刷出来。

lsl330, 2020-03-29:
minimal固件，会出现上述情况，27号固件特地测试了两天。情况和楼上一样

pwrliang, 2020-03-29:
您好，我感觉是lean的源有问题。我把feeds都换成了openwrt官方repo的19.07分支，然后只拷贝需要的lean的package（ssr-plus, softether），编译出的固件已经稳定跑了8天无死机重启。

stevenlinfeng, 2020-03-30:
minimal固件，也是这个情况24日的固件

yanjun42637, 2020-03-31:
目前26日的minimal固件，已经加入zabbix监控，偶尔出现lan口ip不能进入管理界面

monsterdior, 2020-03-31:
27日的固件，有的时候重启之后没办法访问系统，路由正常，lan端访问不了，再次重启之后就会好，持续运行4天没有问题

klever1988, 2020-03-31:
> 27日的固件，有的时候重启之后没办法访问系统，路由正常，lan端访问不了，再次重启之后就会好，持续运行4天没有问题

lienol版本吗？

ft5555, 2020-04-01:
R2S-Lean-2020-03-28固件
 
同样的现象：在 网络 -> 接口 中点击 wan 的 连接 按钮，触发重新连接 PPPoE，等待 1 - 2s IPv4 就会崩溃

只能重启解决

monsterdior, 2020-04-01:
> > 27日的固件，有的时候重启之后没办法访问系统，路由正常，lan端访问不了，再次重启之后就会好，持续运行4天没有问题
> 
> lienol版本吗？

minimal的固件，lean

lukekong, 2020-04-03:
谢谢大佬的付出！0331 lienol版本不死机，Lean版以及Minimal版本经常过一个晚上就死机了。

CsHeng, 2020-04-03:
> 谢谢大佬的付出！0331 lienol版本不死机，Lean版以及Minimal版本经常过一个晚上就死机了。

minimal，丢公司用，基本上第二天过来就凉了

没稳定固件之前不敢扔家里 = =

准备试用下0331 lienol

ft5555, 2020-04-03:
今天更新了R2S-Lean-2020-04-01

在 网络 -> 接口 中点击 wan 的 连接 按钮，触发重新连接 PPPoE

试了几次，没有再崩溃。

lukekong, 2020-04-03:
刚刚更新R2S-Minimal-2020-04-02-ROM

和楼上兄台一样，在 网络 -> 接口 中点击 wan 的 连接 按钮，触发重新连接 PPPoE。试了几次，没有再崩溃。

明天再看看

lukekong, 2020-04-04:
R2S-Minimal-2020-04-02-ROM
等了一天后，发现死机了

15416769, 2020-11-08:
这个问题解决了吗？issue close了，但是我的R2S刷最新的minimal版本还是会过个几天到一周左右就死机了，非得下电再重新上电开机才行