stupidloud/nanopi-openwrt #96: Minimal 版是否可以增加常规的VPN协议（如：L2TP、IKEv、IPsec…… 服务端）以便于通过这些协议可以打通和家里的网络

g1c1img0, 2020-03-28:
Minimal 版是否可以增加常规的VPN协议（如：L2TP、IKEv、IPsec…… 服务端）以便于通过这些协议可以打通和家里的网络？FRP 虽然可以实现内网穿透，但毕竟需要一个中转服务器；对于有公网 IP 的用户来说还是有一定成本且不方便。

(*^__^*)

klever1988, 2020-03-28:
预装了kmod-tun，这些你可以试试用opkg来加

g1c1img0, 2020-03-29:
尝试过命令行安装，发现会报有些包依赖不存在（以 l2tp 为例）

> opkg install xl2tpd
> Collected errors:
> * satisfy_dependencies_for: Cannot satisfy the following dependencies for xl2tpd:
>  * 	kmod-pppol2tp
> * opkg_install_cmd: Cannot install package xl2tpd.

确认了下，软件源中确实不存 `kmod-pppol2tp` 这个包

klever1988, 2020-03-29:
softether socks v2ray socks也都可以回家啊

g1c1img0, 2020-03-29:
嗯 这些确实都可以实现；但对于移动端尤其是 Apple 系列来说都是需要安装第三方客户端的（，不太方便。
没关系的，我再折腾折腾吧；感谢

klever1988, 2020-03-29:
你fork项目，然后修改
https://github.com/klever1988/nanopi-openwrt/blob/master/.github/workflows/r2s_lienol_from_lean.yml#L61

比如ipsec，加入
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y

这样就可以的

fanteral, 2020-03-29:
能否默认增加到固件里面，需要这个呀，命令安装kmod-pppol2tp依赖一直不成功

fanck0605, 2020-03-30:
`src/gz openwrt_core https://downloads.openwrt.org/releases/19.07.2/targets/sunxi/cortexa53/packages`

fanteral, 2020-03-30:
收到，非常感谢

closed 2020-03-31 by klever1988