stupidloud/nanopi-openwrt #111: 机型R2S，请问能否在固件中加入Docker CE？谢谢

djun, 2020-03-31:
你好，我的机型R2S，已刷入2020-3-25 release的Lienol版，需要用到Docker但发现其中没有包含Docker CE，
查资料 官方原版的FriendlyWrt好像是有包含Docker CE的（内核支持+“服务”中有Docker入口），使用Docker是需要在内核编译时加入与之相关的支持的，但研究不出如何修改workflow文件中的编译选项，

请问能否把FriendlyWrt原有的Docker CE加回Lienol版，或其他带passwall的版本？非常感谢！

参考：https://github.com/coolsnowwolf/lede/issues/3334

klever1988, 2020-03-31:
加了内核参数  试试看能不能加入dockerman

klever1988, 2020-03-31:
[image]
你要的docker来了

djun, 2020-03-31:
非常感谢！

djun, 2020-04-01:
> [image]
> 你要的docker来了

[https://github.com/klever1988/nanopi-openwrt/releases/tag/R2S-Lienol-2020-03-31](url)
请问“自动发布 R2S-Lienol-2020-03-31”这个release已经加入的吗？还是需要再等新一天编译的版本？谢谢

klever1988, 2020-04-01:
31加了的 我下来刷过了

djun, 2020-04-01:
好的，先谢谢，晚上测试拉取运行镜像有没有什么问题

maryjeck, 2020-04-01:
R2S的Minimal版本考虑增加么？？

klever1988, 2020-04-01:
> R2S的Minimal版本考虑增加么？？

不考虑 因为docker引擎消耗大量资源

closed 2020-04-01 by klever1988

sjqboss, 2021-09-11:
你的一系列R2S固件SSR+插件有bug，没有订阅功能，请检查。