stupidloud/nanopi-openwrt #116: 能否在编译时加入zerotier

CsHeng, 2020-04-01:
opkg install zerotier失败，报错如下

```
root@FriendlyWrt:~# opkg install zerotier
Installing zerotier (1.4.6-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/zerotier_1.4.6-1_aarch64_cortex-a53.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for zerotier:
 * 	libstdcpp6
 * opkg_install_cmd: Cannot install package zerotier.
root@FriendlyWrt:~# opkg install libstdcpp6
Unknown package 'libstdcpp6'.
Collected errors:
 * opkg_install_cmd: Cannot install package libstdcpp6.
```

klever1988, 2020-04-01:
lean分支版本和lienol分支版本不是有吗

CsHeng, 2020-04-01:
> lean分支版本和lienol分支版本不是有吗

想用上minimal版本  = =

这样子，但竟然另外两个版本有的话，我试试自己从另外两个版本copy一份来编译

closed 2020-04-01 by CsHeng

klever1988, 2020-04-01:
libstdcpp6无法安装可以考虑搭配--nodeps或者--force-depends来强制安装试试

opkg install --nodeps zerotier

CsHeng, 2020-04-01:
> libstdcpp6无法安装可以考虑搭配--nodeps或者--force-depends来强制安装试试
> 
> opkg install --nodeps zerotier

这个可以，非常感谢，我想可以先命令行用着了，有空再折腾编译集成

CsHeng, 2020-04-01:
> libstdcpp6无法安装可以考虑搭配--nodeps或者--force-depends来强制安装试试
> 
> opkg install --nodeps zerotier

用的这个命令安装，可以装，但启动的时候还是报错了。。。

```
root@FriendlyWrt:~# opkg install --force-depends zerotier
Package zerotier (1.4.6-1) installed in root is up to date.
root@FriendlyWrt:~# service zerotier start
Generate secret - please wait...
Error loading shared library libminiupnpc.so.17: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error loading shared library libnatpmp.so.1: No such file or directory (needed by /usr/bin/zerotier-idtool)
Error relocating /usr/bin/zerotier-idtool: UPNP_GetSpecificPortMappingEntry: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_AddPortMapping: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetValidIGD: symbol not found
Error relocating /usr/bin/zerotier-idtool: initnatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: freeUPNPDevlist: symbol not found
Error relocating /usr/bin/zerotier-idtool: closenatpmp: symbol not found
Error relocating /usr/bin/zerotier-idtool: UPNP_GetExternalIPAddress: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendpublicaddressrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: getnatpmprequesttimeout: symbol not found
Error relocating /usr/bin/zerotier-idtool: sendnewportmappingrequest: symbol not found
Error relocating /usr/bin/zerotier-idtool: upnpDiscoverAll: symbol not found
Error relocating /usr/bin/zerotier-idtool: readnatpmpresponseorretry: symbol not found
```

还得装其他工具。。。

想了想我之前直接去下载了zerotier的ipk

然后一步步安装其他依赖，最终应该是这么几个包

```
opkg install libstdcpp6_8.4.0-2_aarch64_cortex-a53.ipk
opkg install --force-overwrite libminiupnpc_2.1.20191224-1_aarch64_cortex-a53.ipk
opkg install libnatpmp_20150609-1_aarch64_cortex-a53.ipk
opkg install zerotier_1.4.6-2_aarch64_cortex-a53.ipk
```

klever1988, 2020-04-01:
那能用么

CsHeng, 2020-04-01:
> 那能用么

能，这样安装完这几个包之后，可以在命令行正常以 `service zerotier start` 启动使用，估计再装个 `luci-app-zerotier` 也应该能用gui了

friskfly, 2020-04-01:
@CsHeng 求指教，这些包在哪里下载的？
opkg install libstdcpp6_8.4.0-2_aarch64_cortex-a53.ipk
opkg install --force-overwrite libminiupnpc_2.1.20191224-1_aarch64_cortex-a53.ipk
opkg install libnatpmp_20150609-1_aarch64_cortex-a53.ipk
opkg install zerotier_1.4.6-2_aarch64_cortex-a53.ipk

klever1988, 2020-04-01:
我加进去吧

CsHeng, 2020-04-02:
> @CsHeng 求指教，这些包在哪里下载的？
> opkg install libstdcpp6_8.4.0-2_aarch64_cortex-a53.ipk
> opkg install --force-overwrite libminiupnpc_2.1.20191224-1_aarch64_cortex-a53.ipk
> opkg install libnatpmp_20150609-1_aarch64_cortex-a53.ipk
> opkg install zerotier_1.4.6-2_aarch64_cortex-a53.ipk

https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/

可以在这里下载到你需要的文件

我是下载了zerotier.ipk尝试安装，然后一步步提示错误，再找对应的包来装，最后总结出来应该是这么几个ipk

不过如果大佬直接加进去的话，那么就简单了