stupidloud/nanopi-openwrt #123: 无线重启以后无法使用，提示“无线未关联”

C-Li, 2020-04-02:
固件：自动发布 R1S-Lean-2020-04-01

我用R1S的无线连接到手机热点，刚开始可以使用，但是点击重启无线以后就用不了了，断电重启也用不了

重启部分的日志如下：
```
Wed Apr  1 23:41:16 2020 daemon.notice wpa_supplicant[17460]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Apr  1 23:41:16 2020 daemon.notice wpa_supplicant[17460]: wlan0: CTRL-EVENT-TERMINATING
Wed Apr  1 23:41:16 2020 daemon.notice netifd: Interface 'wwan' is disabled
Wed Apr  1 23:41:16 2020 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Wed Apr  1 23:41:16 2020 user.notice root: mac80211.sh->mac80211_prepare_vif phy=phy0 ifname=wlan0
Wed Apr  1 23:41:16 2020 daemon.notice netifd: radio0 (23181): command failed: No such device (-19)
Wed Apr  1 23:41:16 2020 user.notice mac80211: Failed command: iw phy phy0 interface add wlan0 type managed
Wed Apr  1 23:41:16 2020 kern.info kernel: [ 1044.799006] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Apr  1 23:41:16 2020 daemon.notice wpa_supplicant[23232]: Successfully initialized wpa_supplicant
Wed Apr  1 23:41:17 2020 daemon.notice netifd: Interface 'wwan' is enabled
Wed Apr  1 23:41:18 2020 daemon.notice wpa_supplicant[23233]: wlan0: Trying to associate with 42:35:2a:8a:b0:32 (SSID='AhWei' freq=2437 MHz)
Wed Apr  1 23:41:18 2020 daemon.notice wpa_supplicant[23233]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
Wed Apr  1 23:41:18 2020 daemon.notice wpa_supplicant[23233]: wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
```

klever1988, 2020-04-02:
删掉接口重新配置把

C-Li, 2020-04-02:
删掉接口重新配置可以用了，重启无线也没问题，但是如果重启路由器又会变成接口未关联，之前用官方的那个固件也有类似的问题，可能硬件有问题吧

klever1988, 2020-04-02:
嗯 r2s上无线也是这样的 无解

closed 2020-04-02 by klever1988