stupidloud/nanopi-openwrt #124: 希望内置软件包easytether-usb-openssl_0.8.9-4_xxx_24kc.ipk

niukai, 2020-04-02:
希望内置软件包：easytether-usb-openssl，我查看了驱动页面：
http://www.mobile-stream.com/easytether/drivers.html
并在此处下载了zip：OpenWrt 19.07+ (instructions): -openssl 
zip链接：
http://www.mobile-stream.com/beta/openwrt/easytether-usb-openssl_0.8.9-5_openwrt-19.07.0.zip
打开zip寻找对应平台的ipk，但是不知道R2S对应是哪个平台？还是zip包中没有R2S平台的ipk？
期望您的解答，谢谢：）
[image]

klever1988, 2020-04-02:
这是干啥的？

niukai, 2020-04-02:
手机开发者模式连接openwrt实现共享手机网络的，这个连接比较方便而且能把Android手机的网络完全映射到openwrt上保持ip一致：）

klever1988, 2020-04-02:
他们好像没有发布源代码啊

niukai, 2020-04-02:
没发布源代码啊？那您知道他们发布的zip包里哪个目录下的ipk对应的是r2s可用的啊？

hackkyo, 2020-04-03:
easytether-usb-openssl_0.8.9-5_openwrt-19.07.0.zip\19.07.0\sunxi\cortexa53

closed 2020-04-03 by klever1988