stupidloud/nanopi-openwrt #143: 在线升级有问题

wangchll, 2020-04-04:
opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz, wget returned 8.

下载依赖的包地址都是404，无法升级

klever1988, 2020-04-04:
[image]

26号以后的固件可以不用执行前两条安装zstd的命令
因为那以后的固件都自带zstd软件了

wangchll, 2020-04-04:
> [image]
> 
> 26号以后的固件可以不用执行前两条安装zstd的命令
> 因为那以后的固件都自带zstd软件了

我试试

closed 2020-04-04 by wangchll