stupidloud/nanopi-openwrt #178: Lean版版系统日志

lhf19900402, 2020-08-25:
Lean版版系统日志里一直有ping8.8.8.8等ip超时的情况，是不是负载均衡里配置的？能否全部换成国内ip?

klever1988, 2020-08-25:
是watchdog插件吧

lhf19900402, 2020-08-25:
我截了个图：
Tue Aug 25 21:00:00 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:00:39 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:01:08 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:01:47 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:02:05 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:02:23 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:03:02 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:03:31 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:03:38 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:03:40 2020 user.info mwan3track[6195]: Check (ping) failed for target "208.67.222.222" on interface wan (pppoe-wan)
Tue Aug 25 21:04:03 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:04:11 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:04:13 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:04:15 2020 user.info mwan3track[6195]: Check (ping) failed for target "208.67.222.222" on interface wan (pppoe-wan)
Tue Aug 25 21:04:20 2020 user.info mwan3track[6195]: Lost 3 ping(s) on interface wan (pppoe-wan)
Tue Aug 25 21:04:33 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:04:35 2020 user.info mwan3track[6195]: Check (ping) failed for target "208.67.222.222" on interface wan (pppoe-wan)
Tue Aug 25 21:04:58 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:05:26 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:05:33 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:05:46 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:05:53 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:06:06 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:06:13 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:06:15 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:06:27 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:06:35 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:06:42 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.8.8" on interface wan (pppoe-wan)
Tue Aug 25 21:07:11 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)
Tue Aug 25 21:07:28 2020 user.info mwan3track[6195]: Check (ping) failed for target "8.8.4.4" on interface wan (pppoe-wan)

lhf19900402, 2020-08-25:
如果我尝试换成 114.114.114.114 系统日志是不再报失败了， 但是路由器就上不了网了  不知道怎么回事。。。
[image]
[image]

[image]

klever1988, 2020-08-25:
你去系统 服务 把mwan相关服务全部停掉就不会弹日志了

closed 2020-08-26 by klever1988