stupidloud/nanopi-openwrt #180: 希望出一个lean稳定版，远程上网课使用

Howardnm, 2020-08-27:
希望出一个lean大更稳定版，精简轻便的，远程上网课用，留学生之痛，只要机器一直稳定运行24小时以上，顺利出国看直播网课，中途不会突然失灵就行了

klever1988, 2020-08-27:
minimal没有问题，lean版东西比较多，稳定性就难说咯

Howardnm, 2020-08-27:
minimal好像没上网插件吧

klever1988, 2020-08-27:
上网插件？

Howardnm, 2020-08-27:
R2S minimal新版的有ssr+吗？

Howardnm, 2020-08-27:
我现在在用lean很不稳定，上着网课突然就挂了

klever1988, 2020-08-27:
> R2S minimal新版的有ssr+吗？

有的, ssrp, passwall和openclash

Howardnm, 2020-08-27:
minimal哪版本最稳定？上课用

klever1988, 2020-08-27:
基本上每一版minimal我都可以运行到下次升级重启，我大概三四天测试一次升级

iamid0, 2020-08-28:
Can we have WireGuard in the mini release?

klever1988, 2020-08-28:
> Can we have WireGuard in the mini release?

you can simply compile your binary with adding CONFIG_PACKAGE_luci-app-wireguard to the configuration area.

iamid0, 2020-08-28:
> > Can we have WireGuard in the mini release?
> 
> 
> 
> you can simply compile your binary with adding CONFIG_PACKAGE_luci-app-wireguard to the configuration area.

Thanks. I will have a try. ^_^

closed 2020-09-08 by klever1988