stupidloud/nanopi-openwrt #184: 8821CU 无法识别

wizcas, 2020-08-30:
首先感谢大佬提供的系统，超级好用。

我这边的设备是R2S，想用无线AP功能，看到说支持 RTL8821CU，于是买了这个：

[image]

用的系统版本是 08-27-Lean，但是设备插上以后不能正确识别：

[image]

于是我想找驱动自己装 [驱动repo](https://github.com/whitebatman2/rtl8821CU)，但是在 `make` 时报错找不到 `/lib/modules/5.4.50/build` （软链到的路径不存在： `/home/runner/work/nanopi-openwrt/nanopi-openwrt/friendlywrt-rk3328/kernel`）。

我对这部分是真的完全不了解，实在不知道应该怎么解决，只能求助于您了。🙏谢谢！

klever1988, 2020-08-30:
插入网卡的时候，观察dmesg打印的三个参数，
[image]

然后编辑/etc/hotplug.d/usb/31-usb_wifi，把那三个参数照猫画虎加进去，就是添加一个if判断，然后拔下来再插看看

wizcas, 2020-08-30:
@klever1988 没有/etc/hotplug.d目录...😓️

wizcas, 2020-08-30:
感谢回复。刚试了一下，wifi应该是可以识别了，因为可以扫描到其他SSID。但是不知道为什么在系统中无法正常启用AP模式，在无线设置那里始终显示 `无线未开启或未关联`。虽然已经选择了AP模式，但是在设置和概况里还是 Client 模式。

我的设置如下：

[image]

在无线设置里显示：

[image]

在概况里显示：

[image]

klever1988, 2020-08-30:
https://github.com/klever1988/nanopi-openwrt/blob/master/JUNK/fix_wifi.sh

执行这个文件 试试看

wizcas, 2020-08-31:
搞定了，多谢。另外发现openwrt里国家选美国，信道2.4G选11，5G选149之后才可以正常设置。不知道什么鬼。。

closed 2020-08-31 by wizcas

klever1988, 2020-08-31:
这个好像是因为新版的op不选国家就没法用无线

klever1988, 2020-08-31:
你把那第三个值给我  或者你把你改的直接提交到友善的源码 https://github.com/friendlyarm/friendlywrt  这样别人编译的固件也能支持你这种网卡了

wizcas, 2020-09-01:
@klever1988 那个文件我后来没改哈，因为看错了，好像pid应该是`C820`。之前那个别的PID应该是蓝牙或内置存储的东西。😓