stupidloud/nanopi-openwrt #201: ssr插件貌似有问题?.....

yuban10703, 2020-09-12:
[image]
[image]

[image]

内网主机列表填写的是一个不存在的ip
但是v2ray服务端的log一直在刷访问www.baidu.com

klever1988, 2020-09-12:
可能是adbyby插件在作妖
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-adbyby-plus/root/usr/share/adbyby/adbybyupdate.sh

yuban10703, 2020-09-12:
emm,不过我没开启adbyby..,这个脚本不开启也会运行么....
[image]

klever1988, 2020-09-12:
不是adbyby
应该是/usr/bin/ssr-switch
这是自动检测和切换失效节点的脚本

yuban10703, 2020-09-12:
emm,我把自动切换关闭了,后台还是在刷,用passwall貌似没问题,挺奇怪的,,,,,
[image]

[image]
下面是ssrp的设置
[image]
[image]
[image]
[image]
[image]
[image]

klever1988, 2020-09-12:
你把/usr/bin/ssr-switch删除 然后重启 看看还刷不刷

yuban10703, 2020-09-12:
我试试

yuban10703, 2020-09-12:
[image]
删除后重启了,然后开机打开ssrp后还在刷......

klever1988, 2020-09-12:
备份一下ssr-check然后删掉它试试

yuban10703, 2020-09-12:
[image]
正常了,emmm
[image]

closed 2020-09-12 by klever1988