stupidloud/nanopi-openwrt #210: 請問R2S哪一個版本有USB外插WIFI功能

eric1975999, 2020-09-16:


klever1988, 2020-09-17:
都有

eric1975999, 2020-09-17:
你好，不好意思我是個小白，第一次使用軟路由，還請多指教，謝謝。我裝了固件後就是找不到可以設定wifi的地方?
[cid:E1F92390-B197-439B-8244-2039BEAE714D-L0-001][cid:690DAB0D-BADB-4775-AFBD-F5E60637B5CA-L0-001]
发自我的iPhone

在 2020年9月17日，21:15，hsu eric <eric.hsu@hotmail.com> 写道：

﻿[cid:16%253aX%253aRgAAAACFrRWuJ1lWTqsmUcY21K7QBwDeJJNUrWPdT5LEqK9kz9iIAAAAAAEJAADeJJNUrWPdT5LEqK9kz9iIAAPOpgfAAAAJ%253aEACJjMJkwR9MRKhy89%252b6LsU6][cid:16%253aX%253aRgAAAACFrRWuJ1lWTqsmUcY21K7QBwDeJJNUrWPdT5LEqK9kz9iIAAAAAAEJAADeJJNUrWPdT5LEqK9kz9iIAAPOpgfAAAAJ%253aEACfOuUlSkczTbJPwVLdOQ3u]

klever1988, 2020-09-17:
只要运行时插入usb网卡，web管理就会出现无线设置，如果没有可能是你的网卡不被支持

提供一下你的usb网卡型号看看吧

eric1975999, 2020-09-17:
明白，网卡型號是騰达W311MA。

发自我的iPhone

klever1988, 2020-09-17:
查过这个usb网卡是mt7601u芯片，我会在下一个版本的“原生测试固件”中加入这个网卡驱动，到时再请测试是否支持

eric1975999, 2020-09-17:
明白、感謝🙏

发自我的iPhone

jungelaile, 2022-07-24:
> 只要运行时插入usb网卡，web管理就会出现无线设置，如果没有可能是你的网卡不被支持
> 
> 提供一下你的usb网卡型号看看吧

有支持MT7612U型号的驱动支持吗,谢谢