stupidloud/nanopi-openwrt #220: R2S在原生校园网的环境下怎么设置ipv6？

CrazyHsu, 2020-09-23:
楼主大大您好，我是一个在校学生，有着原生的校园ipv6环境，在网上购入已经刷好了friendlyWRT（版本：OpenWrt R20.7.1 / LuCI Master (git-20.223.26773-d18ef13) built by DYC ）的R2S，但是通上电lan口插上电脑以后，电脑并没有获取到ipv6地址，于是我把网线直接插到电脑上，电脑又是可以获取到ipv6地址的，请问有什么办法可以解决吗？感谢了！
下面是把网线插在R2S上的图，可以看到没有获取到ipv6
[image]

klever1988, 2020-09-23:
网上有很多教程 试试看吧
https://haoyu.love/blog646.html

Chenxin9607, 2020-09-25:
我也是校园网，用上面的连接配置也不是很好用，https://www.mereith.com/2018/12/06/ipv6%E8%BD%AC%E5%8F%91/，这个主页的教程，中继模式成功了。

Chenxin9607, 2020-09-25:
我也是校园网用上面的连接不是很好用，使用下面这个链接的中继模式成功了

https://www.mereith.com/2018/12/06/ipv6%E8%BD%AC%E5%8F%91/