stupidloud/nanopi-openwrt #231: DDNS服务供应商不会刷新

zhoujian1898, 2020-10-01:
R2S-Lean-2020-09-30-ROM版本, 在软件包安装ddns-scripts_no-ip_com, 版本2.8.0-24, 成功后转到DDNS界面修改DDNS配置, 在服务提供商选项那无法看到安装的no-ip的选项. 重启R2S后也无法刷新出no-ip的选项.
Remark: 更新ddns-scripts到2.8版本后, 服务提供商选项只剩自定义, 无法看到任何提供商的选项.

ddcatgg, 2020-10-03:
借楼提问：其实通过cloudflare api也可以实现DDNS，不知是否能实现并提供cloudflare的ddns选项？

    https://api.cloudflare.com/client/v4/zones/{zone_id}/dns_records/{dns_record_id}

hxlls, 2020-10-05:
> 借楼提问：其实通过cloudflare api也可以实现DDNS，不知是否能实现并提供cloudflare的ddns选项？
> 
> ```
> https://api.cloudflare.com/client/v4/zones/{zone_id}/dns_records/{dns_record_id}
> ```

其实是可以的，只是没勾选而已
[image]

ddcatgg, 2020-10-06:
> 其实是可以的，只是没勾选而已
那我自己有办法勾选么？还有目前除了站长，其他人有没有办法启动action自动编译呢？谢谢！

zlf179, 2020-10-06:
> > 其实是可以的，只是没勾选而已
> 
> 那我自己有办法勾选么？还有目前除了站长，其他人有没有办法启动action自动编译呢？谢谢！

可以 我就fork到自己仓库 action编译成功

ddcatgg, 2020-10-07:
> 可以 我就fork到自己仓库 action编译成功

求启动这些action的方法