stupidloud/nanopi-openwrt #244: RTL8821CU的驱动如何在openwrt上正常使用的

rote66, 2020-10-12:
RT,在别的开发板上测试都是只能扫描到wifi，但是对它进行网络操作就无限卡死。。
openwrt官方几个月前貌似加了驱动进去，很快有人提了issue，但是也很快就被打上了wontfix标签。。不知道是不是和openwrt的mac80211不兼容？
貌似全网就只看到nanopi上正常用
故想求助咨询下dalao

klever1988, 2020-10-14:
i am not dalao
目前仅知道mac80211版本为5以上时无线会卡死，所以我一直带4.x版本的op分支进行编译

rote66, 2020-10-15:
> i am not dalao
> 目前仅知道mac80211版本为5以上时无线会卡死，所以我一直带4.x版本的op分支进行编译

唔，我尝试使用lienol大佬的openwrt源码进行编译，里面的mac80211是4.19版本的，然而还是出现了卡死的问题。
不知道是mac80211的锅还是驱动的锅
我看你们的changelog，在3月12日加入了对8821cu的支持，但是查看那时候的变更并没有看到config里写了8821cu的东西。而是在9月21日的才看到了kmod-rtl8821cu字样，然后又看到nanopir2s好像是官方写的支持8821cu这块卡，所以这个8821cu的支持是friendlywrt里面带的驱动么

klever1988, 2020-10-16:
> 唔，我尝试使用lienol大佬的openwrt源码进行编译，里面的mac80211是4.19版本的，然而还是出现了卡死的问题。
> 不知道是mac80211的锅还是驱动的锅 [...]

https://github.com/friendlyarm/sd-fuse_rk3328/commit/f2f556de9ce0f74f1464c0ec7ca28872f6dbbdd8
是友善给linux内核打入驱动才得到支持
你可以试试用minimal刷入看能不能用你的网卡建立热点，如果还是卡死，可能是网卡的问题了

rote66, 2020-10-17:
> > > i am not dalao
> > > 目前仅知道mac80211版本为5以上时无线会卡死，所以我一直带4.x版本的op分支进行编译
> > 
> > 
> > 唔，我尝试使用lienol大佬的openwrt源码进行编译，里面的mac80211是4.19版本的，然而还是出现了卡死的问题。
> > 不知道是mac80211的锅还是驱动的锅
> > 我看你们的changelog，在3月12日加入了对8821cu的支持，但是查看那时候的变更并没有看到config里写了8821cu的东西。而是在9月21日的才看到了kmod-rtl8821cu字样，然后又看到nanopir2s好像是官方写的支持8821cu这块卡，所以这个8821cu的支持是friendlywrt里面带的驱动么
> 
> [friendlyarm/sd-fuse_rk3328@f2f556d](https://github.com/friendlyarm/sd-fuse_rk3328/commit/f2f556de9ce0f74f1464c0ec7ca28872f6dbbdd8)
> 是友善给linux内核打入驱动才得到支持
> 你可以试试用minimal刷入看能不能用你的网卡建立热点，如果还是卡死，可能是网卡的问题了

网卡我确认是没有问题的，debian/ubuntu是可以正常用的。
配置都不一样刷不了minimal的吧
friendlyarm官方那份fork的rtl8821cu驱动，单独编译无法insmod，带上mac80211编译不过。
github上找过一份不带mac80211编译驱动会引发崩溃。带上又卡死。
可能还是和openwrt的mac80211不兼容吧
不清楚为何友善编译的能正常加载进，而且脱离80211