stupidloud/nanopi-openwrt #246: Aliyun的DDNS无法成功更新IP, 求大神帮忙看看

zhoujian1898, 2020-10-13:
Rom版本: R2S-Lean-2020-10-12-ROM, R2S-Lean-2020-10-10-ROM
在DDNS服务中, 选择Aliyun供应商, 填写好信息好, IP地址一直无法更新到DNS服务器中. 无法确定问题点.
DDNS Log如下:
204019       : verbose mode  : 0 - run normal, NO console output
 204019       : check interval: 600 seconds
 204019       : force interval: 259200 seconds
 204020       : retry interval: 60 seconds
 204020       : retry counter : 0 times
 204020       : No old process
 204020       : last update: never
 204020       : Detect registered/public IP
 204020       : #> /usr/bin/nslookup zhoujian1898.top  >/var/run/ddns/WAN_DDNS.dat 2>/var/run/ddns/WAN_DDNS.err
 204020  WARN : NO valid IP found
 204020  info : Starting main loop at 2020-10-13 20:40
 204021       : Detect local IP on 'network'
 204021       : Local IP '114.216.102.45' detected on network 'wan'
 204021       : Update needed - L: '114.216.102.45' <> R: ''
 204021       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 204021       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/WAN_DDNS.dat -o /var/run/ddns/WAN_DDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=zhoujian1898.top&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A40%3A21Z&SignatureVersion=1.0&SignatureNonce=f203100f-ee1c-4615-9597-e776d4373f53&Signature=76i8WsHXKqBNuX9E7XruC2nxNTs%3D'
 204023       : 获取到解析记录: {"TotalCount":0,"RequestId":"71571170-09C6-4E91-9D05-C8F0CC5E7A9C","PageSize":20,"DomainRecords":{"Record":[]},"PageNumber":1}
 204023       : 解析记录不存在
 204027       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/WAN_DDNS.dat -o /var/run/ddns/WAN_DDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A40%3A26Z&SignatureVersion=1.0&SignatureNonce=729e7946-faa8-43d3-8385-fbc385b32f59&Signature=twlGw1dgWuW0gtNPGhJhIg%2BotAw%3D'
 204028 ERROR : wget 错误代码: '8'
 204028       : http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A40%3A26Z&SignatureVersion=1.0&SignatureNonce=729e7946-faa8-43d3-8385-fbc385b32f59&Signature=twlGw1dgWuW0gtNPGhJhIg%2BotAw%3D:
2020-10-13 20:40:28 ERROR 400: Bad Request.
 204029  WARN : 传输失败 - 1/0 在 60 秒后重试
 204130       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/WAN_DDNS.dat -o /var/run/ddns/WAN_DDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A41%3A29Z&SignatureVersion=1.0&SignatureNonce=67e523fa-f8c0-42f2-8e75-bafdb76f3568&Signature=rfEL1mujy3tyAaW0pncXroG3MfU%3D'
 204131 ERROR : wget 错误代码: '8'
 204131       : http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A41%3A29Z&SignatureVersion=1.0&SignatureNonce=67e523fa-f8c0-42f2-8e75-bafdb76f3568&Signature=rfEL1mujy3tyAaW0pncXroG3MfU%3D:
2020-10-13 20:41:31 ERROR 400: Bad Request.
 204132  WARN : 传输失败 - 2/0 在 60 秒后重试
 204233       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/WAN_DDNS.dat -o /var/run/ddns/WAN_DDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A42%3A32Z&SignatureVersion=1.0&SignatureNonce=d79a7523-9a57-49f3-9cdc-454ad5253856&Signature=kfcvz%2FW%2BMe3AHBsWsgtgomtETE4%3D'
 204234 ERROR : wget 错误代码: '8'
 204234       : http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A42%3A32Z&SignatureVersion=1.0&SignatureNonce=d79a7523-9a57-49f3-9cdc-454ad5253856&Signature=kfcvz%2FW%2BMe3AHBsWsgtgomtETE4%3D:
2020-10-13 20:42:34 ERROR 400: Bad Request.
 204235  WARN : 传输失败 - 3/0 在 60 秒后重试
 204336       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/WAN_DDNS.dat -o /var/run/ddns/WAN_DDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A43%3A36Z&SignatureVersion=1.0&SignatureNonce=09ce4cfb-82f2-46e6-bad5-8a602f9b26fb&Signature=R0aLfVMtBxYe5%2BgzagS2P90eUjE%3D'
 204337 ERROR : wget 错误代码: '8'
 204337       : http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=top&RR=zhoujian1898&Type=A&Value=114.216.102.45&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAI4G1DKVKEjgLquB654JSB&SignatureMethod=HMAC-SHA1&Timestamp=2020-10-13T12%3A43%3A36Z&SignatureVersion=1.0&SignatureNonce=09ce4cfb-82f2-46e6-bad5-8a602f9b26fb&Signature=R0aLfVMtBxYe5%2BgzagS2P90eUjE%3D:
2020-10-13 20:43:37 ERROR 400: Bad Request.
 204338  WARN : 传输失败 - 4/0 在 60 秒后重试

klever1988, 2020-10-14:
你应该先创建一条a记录zhoujian1898.top指向随便一个ip吧， 这插件可能只有更新a记录的ip值，不能自己创建a记录

zhoujian1898, 2020-10-14:
大神牛逼~~
我就说为啥一直不成功呢~~

closed 2020-10-14 by zhoujian1898