stupidloud/nanopi-openwrt #252: opkg update部分报错

bike3097a, 2020-10-16:
BusyBox v1.30.0 () built-in shell (ash)

  ___    _             _ _    __      _____ _____
 | __| _(_)___ _ _  __| | |_  \ \    / / _ \_   _|
 | _| '_| / -_) ' \/ _` | | || \ \/\/ /|   / | |
 |_||_| |_\___|_||_\__,_|_|\_, |\_/\_/ |_|_\ |_|
                           |__/
 -----------------------------------------------------
 FriendlyWRT SNAPSHOT, r2993+618-b9a408c2b4
 -----------------------------------------------------

DYC Build

root@FriendlyWrt:~# opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/helloworld/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/helloworld/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/helloworld/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz, wget returned 8.
root@FriendlyWrt:~#

klever1988, 2020-10-16:
这几个源是不存在的 不用管错误提示 下一步该干啥干啥

closed 2020-10-16 by klever1988

bike3097a, 2020-10-16:
为啥不修复呢？