stupidloud/nanopi-openwrt #256: 今天的usb无线刚到，试了下，不能使用

bike3097a, 2020-10-17:
请尽快解决，

4👈fu植文本₴4D1AcSuVIHe₴转移至👉ta0寶👈或点几链街 https://m.tb.cn/h.4YkN9CA?sm=d0cd5a 至瀏lan嘂..【NanoPi R2S路由器AP USB WIFI无线网卡AC双频OPENWRT免驱直插包邮】

买的这款。。。

bike3097a, 2020-10-17:
id是0bda:c811 具体不清楚是什么芯片

iscodeeasyornot, 2020-11-03:
FriendlyWrt 可编译进 811ac 等 USB 网卡，设置为 N 模式，5GHz。强制频宽 40MHz，信道 40。重启后，可以跑到 100Mbps 左右。

Deckard-f-Cain, 2020-11-03:
> FriendlyWrt 可编译进 811ac 等 USB 网卡，设置为 N 模式，5GHz。强制频宽 40MHz，信道 40。重启后，可以跑到 100Mbps 左右。

我N模式只能2.4G，你其他设置通报一下，让我们学习一下。比如 信道用的多少？国家选的哪个？还有其他所有可勾选项，你勾选了什么？
我只能在legacy模式，149信道，美国条件下开启5G

bike3097a, 2020-11-03:
> FriendlyWrt 可编译进 811ac 等 USB 网卡，设置为 N 模式，5GHz。强制频宽 40MHz，信道 40。重启后，可以跑到 100Mbps 左右。

我的不可以哦

slb116763814, 2020-12-21:
@iscodeeasyornot 我的根本不能切换5GHz,一切换就不能用。

iscodeeasyornot, 2020-12-21:
用的是friendlywrt的话按照我上面发的来应该没问题，8811cu和8821au全部测试通过。8821cu还可以调n和ac两个模式

发自我的iPhone