stupidloud/nanopi-openwrt #265: 麻烦添加smartdns插件

bike3097a, 2020-10-22:
请作者添加smartdns插件，谢谢。

paniy, 2020-10-23:
同样想要

TomFoxLee, 2020-11-11:
同求。安裝自行下載的 smartdns ipk 介面總出不來，不知道是哪裡的問題 :~~~

bike3097a, 2020-11-13:
我已经更换其他家的openwrt系统了

TomFoxLee, 2020-11-13:
後來看到 Leon 大的相關說明才發現，其實 SmartDNS 並不是必要的？！
https://github.com/coolsnowwolf/lede/issues/2551

之前用 Merlin 的時候，覺得 SmartDNS 可以改善連線紅X的問題，
現在換用了 OpenWRT 才知道，它的運作邏輯不是我之前想的那麼回事 @.@

所以還有類似疑問的人可以先不用糾結了，SmartDNS 並不是運作正常的必要條件，內建的 pdnsd 搭配 SSRP 已經足夠了，
SmartDNS 更多的是錦上添花，而且必須是理解怎麼正確設置的前提之下。要不然沒設對反倒添亂。

fushenzhi, 2021-01-16:
> 
> 
> 同求。安裝自行下載的 smartdns ipk 介面總出不來，不知道是哪裡的問題 :~~~

我也是遇到自己安装的smartdns配置页面打不开，不知如何解决。我是旁路由模式启用clash+AdGuard，要么去广告效果不好，要么翻墙失效，要么部分国内网站打开巨慢，配置的好艰难。有没有大佬教我怎么配置这俩总有冲突的插件啊？

fushenzhi, 2021-01-16:
> 
> 
> 後來看到 Leon 大的相關說明才發現，其實 SmartDNS 並不是必要的？！
> [coolsnowwolf/lede#2551](https://github.com/coolsnowwolf/lede/issues/2551)
> 
> 之前用 Merlin 的時候，覺得 SmartDNS 可以改善連線紅X的問題，
> 現在換用了 OpenWRT 才知道，它的運作邏輯不是我之前想的那麼回事 @.@
> 
> 所以還有類似疑問的人可以先不用糾結了，SmartDNS 並不是運作正常的必要條件，內建的 pdnsd 搭配 SSRP 已經足夠了，
> SmartDNS 更多的是錦上添花，而且必須是理解怎麼正確設置的前提之下。要不然沒設對反倒添亂。

我固件R2S做旁路由，配置openclash+AdGuard，想要实现翻墙+去广告(特别是手机端各种app启动页面的广告)，然后遇到BUG两大卡车！配的我手脚发麻！跪求大佬指点一二，如果能推荐一下AdGuard的"上游DNS设置"和去广告规则就太棒了。