stupidloud/nanopi-openwrt #272: 无法可写挂在NTFS格式的硬盘

coolboy0961, 2020-10-24:
路由型号：黄壳R2S
固件版本：R2S-Minimal-2020-10-23

执行mount命令得到如下结果
```
/dev/sdb1 on /mnt/sdb1 type ntfs (ro,relatime,uid=0,gid=0,fmask=0177,dmask=077,nls=utf8,errors=continue,mft_zone_multiplier=1)
/dev/sdc1 on /mnt/sdc1 type ntfs (ro,relatime,uid=0,gid=0,fmask=0177,dmask=077,nls=utf8,errors=continue,mft_zone_multiplier=1)
```

想安装ntfs-3g接连出现若干错误。
1. 安装ntfs-3g提示package size mismatch
```
root@FriendlyWrt:/# opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
root@FriendlyWrt:/# opkg install ntfs-3g
Installing ntfs-3g (2017.3.23-3-fuseint) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk
Collected errors:
 * opkg_install_pkg: Package size mismatch: ntfs-3g is 186565 bytes, expecting 186599 bytes
 * opkg_install_cmd: Cannot install package ntfs-3g.
```

手动下载package然后安装可以解决上述错误
```
root@FriendlyWrt:~# wget https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk
--2020-10-24 14:37:48--  https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages/ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk
Resolving openwrt.proxy.ustclug.org... 192.109.232.118
Connecting to openwrt.proxy.ustclug.org|192.109.232.118|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 186565 (182K) [application/octet-stream]
Saving to: 'ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk'

ntfs-3g_2017.3.23-3-fuseint_aarch64_cor 100%[===============================================================================>] 182.19K  --.-KB/s    in 0.1s    

2020-10-24 14:37:48 (1.28 MB/s) - 'ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk' saved [186565/186565]

opkg install ntfs-3g_2017.3.23-3-fuseint_aarch64_cortex-a53.ipk 
Installing ntfs-3g (2017.3.23-3-fuseint) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ntfs-3g:
 *      kmod-fuse
 * opkg_install_cmd: Cannot install package ntfs-3g.
```

2. 本地安装会提示没有所需的kmod-fuse package。
```
root@FriendlyWrt:~# opkg install kmod-fuse
Unknown package 'kmod-fuse'.
Collected errors:
 * opkg_install_cmd: Cannot install package kmod-fuse.
```

目前卡在这个地方。无法安装ntfs-3g就无法可写的mount ntfs分区，也就无法使用ntfs分区硬盘成为NAS的一部分。
有人解决过类似问题吗？

coolboy0961, 2020-10-24:
将如下源添加到发行版软件源中，执行opkg update。
```
src/gz openwrt_core https://downloads.openwrt.org/releases/19.07.2/targets/sunxi/cortexa53/packages
```

然后尝试安装kmod-fuse，得到如下错误。

```
root@FriendlyWrt:~# opkg install kmod-fuse
Installing kmod-fuse (4.14.171-1) to root...
Downloading https://downloads.openwrt.org/releases/19.07.2/targets/sunxi/cortexa53/packages/kmod-fuse_4.14.171-1_aarch64_cortex-a53.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fuse:
 *      kernel (= 4.14.171-1-87234a97d59467c217fc0de872097ed8)
 * opkg_install_cmd: Cannot install package kmod-fuse.
```

klever1988, 2020-10-24:
opkg install ntfs-3g --nodeps

coolboy0961, 2020-10-24:
@klever1988 

非常感谢！解决了。原来不安装dependency的package也行啊。  
用下面的命令至少能手动挂载rw的ntfs硬盘了。
```
mount -t ntfs-3g /dev/sdc1 /mnt/sdc1
```

closed 2020-10-24 by coolboy0961

coolboy0961, 2020-10-25:
最终在启动项的启动脚本中添加了如下代码实现了开机自动挂载可写的NTFS分区。
```
# mount ntfs-3g device
mkdir -p /mnt/sdb1 && ntfs-3g /dev/sdb1 /mnt/sdb1
mkdir -p /mnt/sdc1 && ntfs-3g /dev/sdc1 /mnt/sdc1
```

fasthold, 2021-02-12:
> opkg install ntfs-3g --nodeps

下个版本考虑直接把ntfs做成默认功能吗？毕竟现在挂个U盘移动硬盘什么的也是基础功能啊