stupidloud/nanopi-openwrt #286: 请教如何通过4G Modem上网

coolboy0961, 2020-10-28:
路由型号：黄壳R2S
固件版本：R2S-Minimal-2020-10-23

想添加一个USB的4G Modem接口，然后用mwan3实现光纤和4G的双WAN上网。
买了一个4G的USB猫，插入R2S后设备显示了。
[image]

然后添加接口的时候，提示未安装协议支持。
[image]

安装comgt的时候提示无法找到。
[image]

是否有大侠可以指点一二？有没有安装comgt的方法？
还是R2S根本无法用4G猫上网？

下面是我买的4G Modem
[image]
[image]

coolboy0961, 2020-10-28:
查找了一下相关资料，要4G modem上网需要安装luci-proto-3g。
找到一个可以用的源，但是不知道是不是可以安心的用，希望老大可以帮忙看看。
https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa53/packages

总之按照下面的方法安装上了。
```
root@FriendlyWrt:~# wget https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa53/packages/comgt_0.32-33_aarch64_cortex-a53.ipk
--2020-10-28 23:31:32--  https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa53/packages/comgt_0.32-33_aarch64_cortex-a53.ipk
Resolving openwrt.proxy.ustclug.org... 192.109.232.118
Connecting to openwrt.proxy.ustclug.org|192.109.232.118|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 21864 (21K) [application/octet-stream]
Saving to: 'comgt_0.32-33_aarch64_cortex-a53.ipk'

comgt_0.32-33_aarch64_cortex-a53.ipk                                           100%[====================================================================================================================================================================================================>]  21.35K  --.-KB/s    in 0.07s

2020-10-28 23:31:34 (289 KB/s) - 'comgt_0.32-33_aarch64_cortex-a53.ipk' saved [21864/21864]

root@FriendlyWrt:~# wget https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base/chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk
--2020-10-28 23:32:54--  https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base/chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk
Resolving openwrt.proxy.ustclug.org... 192.109.232.118
Connecting to openwrt.proxy.ustclug.org|192.109.232.118|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10434 (10K) [application/octet-stream]
Saving to: 'chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk'

chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk                             100%[====================================================================================================================================================================================================>]  10.19K  --.-KB/s    in 0s

2020-10-28 23:32:58 (26.8 MB/s) - 'chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk' saved [10434/10434]

root@FriendlyWrt:~# opkg install chat_2.4.8.git-2020-10-03-2_aarch64_cortex-a53.ipk
Installing chat (2.4.8.git-2020-10-03-2) to root...
Configuring chat.
root@FriendlyWrt:~# opkg install comgt_0.32-33_aarch64_cortex-a53.ipk
Installing comgt (0.32-33) to root...
Configuring comgt.
root@FriendlyWrt:~# opkg install luci-proto-3g
Installing luci-proto-3g (git-20.272.49300-479154e-1) to root...
Downloading https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/aarch64_cortex-a53/luci/luci-proto-3g_git-20.272.49300-479154e-1_all.ipk
Configuring luci-proto-3g.
```

但是还是无法连接成功。
系统日志持续出现下面的log。
```
Wed Oct 28 23:47:09 2020 daemon.notice netifd: Interface '4GMODEM' is setting up now
Wed Oct 28 23:47:11 2020 daemon.notice pppd[7340]: pppd 2.4.8 started by root, uid 0
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: abort on (BUSY)
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: abort on (NO CARRIER)
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: abort on (ERROR)
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: report (CONNECT)
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: timeout set to 10 seconds
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: send (AT&F^M)
Wed Oct 28 23:47:12 2020 local2.info chat[7350]: expect (OK)
Wed Oct 28 23:47:22 2020 local2.info chat[7350]: alarm
Wed Oct 28 23:47:22 2020 local2.info chat[7350]: Failed
Wed Oct 28 23:47:22 2020 daemon.err pppd[7340]: Connect script failed
Wed Oct 28 23:47:23 2020 daemon.info pppd[7340]: Exit.
Wed Oct 28 23:47:23 2020 daemon.notice netifd: Interface '4GMODEM' is now down
```

下面是我的接口设置
[image]

其实应该用电脑连上这个4g modem先看看能不能上网的，可是没有windows PC没法验证。。。
sim卡是没问题的，手机上可以上网，pocket wifi上也可以上网。

coolboy0961, 2020-10-28:
根据下面的介绍，
https://openwrt.org/docs/guide-user/network/wan/wwan/3gdongle

我发现r2s的固件下缺少下面的包，也许这导致了r2s和USB猫之间无法正确通信。
而我就算找到别人编译的包也会因为kernal版本不同而无法安装。
kmod-usb-serial kmod-usb-serial-option kmod-usb-serial-wwan

不知道老大能否告诉一下如何添加这些包的方法？

coolboy0961, 2020-10-29:
@klever1988 

我觉得只要添加正确的core源能安装kmod-usb-serial等包就可以继续了，能告诉我一下正确的源地址吗？
自己试了很多都说kernal版本不对无法安装。
src/gz openwrt_core

coolboy0961, 2020-10-29:
发现已经安装的kmod的kernel版本基本都是4.14.180-1。
从[这里](https://openwrt.proxy.ustclug.org/releases/19.07.3/targets/sunxi/cortexa53/kmods/4.14.180-1-87234a97d59467c217fc0de872097ed8/)找到了同样版本的kmod。
用`opkg install kmod-usb-seria`还是提示kernel版本错误无法安装，添加了一个参数就安装上了`opkg install kmod-usb-serial --force-depends`，不过还是会提示错误，不去管它。

安装的包一览中已经可以看到了。先这么装上试试。
[image]

coolboy0961, 2020-10-29:
这个4G Modem似乎不支持我的sim卡。。。
等换一个Modem确认可以上网以后再继续吧。

coolboy0961, 2020-10-29:
放弃4G Modem改用USB Tethering了。

根据[这里](https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering)的介绍，如果是安卓手机的话，只需要安装一个包。
```
opkg install kmod-usb-net-rndis
```
这个包已经存在与r2s固件当中了。

USB连接上手机并开启USB Tethering后，建立一个名为wanb的接口。
[image]
[image]
[image]

正确连接的话可以看到DHCP分配的地址和流量
[image]

开启WAN防火墙的强制连接跟踪。
[image]

设置wan和wanb的使用网关跃点
[image]
[image]

保存&应用后重启r2s。

然后验证两个网口是否都能连接外网
```
ping -c 1 -I wan www.google.com
ping -c 1 -I wanb www.google.com
```

接下来安装mwan3。
我设置的源
```
src/gz openwrt_core https://openwrt.proxy.ustclug.org/releases/19.07.3/targets/sunxi/cortexa53/kmods/4.14.180-1-87234a97d59467c217fc0de872097ed8
src/gz openwrt_luci https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/aarch64_cortex-a53/luci

src/gz openwrt_base https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/base
src/gz openwrt_packages https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/packages
src/gz openwrt_routing https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/routing
```

```
opkg update
opkg install mwan3
```
mwan3安装时会报错说size不对，直接wget 下载包以后本地安装。
```
opkg install mwan3_2.10.0-1_all.ipk
```
然后安装luci-app-mwan3
```
opkg install luci-app-mwan3
```

安装完后状态和网络下都会出现负载均衡的选项。
[image]
[image]

点击网络的负载均衡，只保留wan和wanb，其他的删除
[image]

点击规则，只保留DEFAULT_RULE_V4，其他的删除
[image]
将分配的策略设置为balanced可以实现同时用wan和wanb上网，默认是wan60% wanb40%
[image]
如果想要调节比例，去成员里添加或修改成员，通过调整越点数和比重，然后去策略里面添加新的策略。
[image]
[image]

设置结束后点击保存&应用，然后重启mwan3
```
mwan3 restart
```

打开状态的负载均衡，可以看到wan和wanb都已被激活。
[image]

detail可以看到负载均衡的比例
[image]

到此终于实现了双wan负载均衡上网。
将详细过程记录于此以便于以后刷机后重新设定时参考，也可以方便一下别的有同样需求的朋友。

closed 2020-10-29 by coolboy0961

coolboy0961, 2020-10-29:
一些参考网站
https://openwrt.org/docs/guide-user/network/wan/wwan/3gdongle
https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols#protocol_3g_ppp_over_ev-do_cdma_umts_or_gprs
https://qiita.com/daredeshow/items/7af368e6b95b62a04e5e
https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering
https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3
https://forum.openwrt.org/t/solved-mwan3-mwan-interface-live-status-showing-offline-for-all-interfaces/3754

BetterLuke, 2020-12-16:
> 放弃4G Modem改用USB Tethering了。
>
> 根据[这里](https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering)的介绍，如果是安卓手机的话，只需要安装一个包。
>
> ```
> opkg install kmod-usb-net-rndis
> ```
>
> 这个包已经存在与r2s固件当中了。 [...]

用不断网！牛