stupidloud/nanopi-openwrt #288: opkg update 失敗

isinglever, 2020-10-31:
我按照readme里的方法進行在線升級，但是第一步就報錯了。請問如何解決呢
`Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/targets/rockchip-rk3328/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/helloworld/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/aarch64_cortex-a53/lienol/Packages.gz, wget returned 8.`

isinglever, 2020-10-31:
這個連接提示404 not found

coolboy0961, 2020-11-01:
多搜搜以前的issue。
https://github.com/klever1988/nanopi-openwrt/issues/286#issuecomment-718837157

isinglever, 2020-11-01:
> 多搜搜以前的issue。
> [#286 (comment)](https://github.com/klever1988/nanopi-openwrt/issues/286#issuecomment-718837157)

感謝解答

closed 2020-11-01 by isinglever