stupidloud/nanopi-openwrt #294: 请问R2S-native版本的能不能继续更新

zjp0703, 2020-11-04:
我使用R2S-native版本的固件，感觉相当稳定，其余版本使用了几个月，关闭IPV6依然有死机现象
R2S-native使用了快一个月了没死机过一次。

klever1988, 2020-11-04:
这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦

https://github.com/project-openwrt/openwrt

bike3097a, 2020-11-04:
> 这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦
> 
> https://github.com/project-openwrt/openwrt
> 
> 加上没有usb无线驱动 所以我才拿掉啦

请问可以自己安装驱动吗？

zjp0703, 2020-11-05:
> 这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦
> 
> https://github.com/project-openwrt/openwrt
> 
> 加上没有usb无线驱动 所以我才拿掉啦

有尝试过自己编译，我用在vm虚拟机下编译，非root用户有些命令执行不了，试过很多编译教程都一样。
大佬要是愿意可以一两个月更新一次这个版本，一个稳定的东西，我们也不会天天去刷机。

bike3097a, 2020-11-05:
> > 这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦
> > https://github.com/project-openwrt/openwrt
> > 加上没有usb无线驱动 所以我才拿掉啦
> 
> 有尝试过自己编译，我用在vm虚拟机下编译，非root用户有些命令执行不了，试过很多编译教程都一样。
> 大佬要是愿意可以一两个月更新一次这个版本，一个稳定的东西，我们也不会天天去刷机。

你直接root账户更新不就好了

bike3097a, 2020-11-07:
> > 这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦
> > https://github.com/project-openwrt/openwrt
> > 加上没有usb无线驱动 所以我才拿掉啦
> 
> 有尝试过自己编译，我用在vm虚拟机下编译，非root用户有些命令执行不了，试过很多编译教程都一样。
> 大佬要是愿意可以一两个月更新一次这个版本，一个稳定的东西，我们也不会天天去刷机。

明天我编译，在安装ubuntu系统

zjp0703, 2020-11-08:
> > > 这个版本的固件挺好编译的 稍微找一下openwrt的编译教程文章就可以自己动手啦
> > > https://github.com/project-openwrt/openwrt
> > > 加上没有usb无线驱动 所以我才拿掉啦
> > 
> > 
> > 有尝试过自己编译，我用在vm虚拟机下编译，非root用户有些命令执行不了，试过很多编译教程都一样。
> > 大佬要是愿意可以一两个月更新一次这个版本，一个稳定的东西，我们也不会天天去刷机。
> 
> 明天我编译，在安装ubuntu系统

大佬，编译好了通知我