stupidloud/nanopi-openwrt #321: 关于Lean固件usb网卡问题

Wyaohui, 2020-12-03:
下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是Client模式，怎么变成master模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？

Leon11111111, 2020-12-03:
> 下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是客户端模式，怎么变成主模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？

你好我多句嘴，无线改为AP模式就是MASTER模式了。

Wyaohui, 2020-12-03:
> > 下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是客户端模式，怎么变成主模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？
> 
> 你好我多句嘴，无线改为AP模式就是MASTER模式了。

你好，就是用的AP模式，固件会显示master，然后会自动切换成client，很奇怪

Leon11111111, 2020-12-03:
> > > 下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是客户端模式，怎么变成主模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？
> > 
> > 
> > 你好我多句嘴，无线替换AP模式就是MASTER模式了。
> 
> 你好，就是用的AP模式，固件会显示master，然后会自动切换成client，很奇怪

不过我遇到的也是WIFI问题，启动了SSR plus+的节点以后手机连接路由器就不正常，你看我前面发的问题。-_-

Wyaohui, 2020-12-03:
> > > > 下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是客户端模式，怎么变成主模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？
> > > 
> > > 
> > > 你好我多句嘴，无线替换AP模式就是MASTER模式了。
> > 
> > 
> > 你好，就是用的AP模式，固件会显示master，然后会自动切换成client，很奇怪
> 
> 不过我遇到的也是WIFI问题，启动了SSR plus+的节点以后手机连接路由器就不正常，你看我前面发的问题。-_-

看到了，问问大佬是什么情况，话说大佬的固件，wan和lan没有互换是么

Leon11111111, 2020-12-03:
> > > > > 下载了大佬的2020-12-02Lean固件，usb网卡是comfast的811ac，进入系统看到了无线的选项，为什么显示一直是客户端模式，怎么变成主模式？还有大佬，为啥一接入wan线，路由器后台就进不去了？
> > > > 
> > > > 
> > > > 你好我多句嘴，无线替换AP模式就是MASTER模式了。
> > > 
> > > 
> > > 你好，就是用的AP模式，固件会显示master，然后会自动切换成client，很奇怪
> > 
> > 
> > 不过我遇到的也是WIFI问题，启动了SSR plus+的节点以后手机连接路由器就不正常，你看我前面发的问题。-_-
> 
> 看到了，问问大佬是什么情况，话说大佬的固件，wan和lan没有互换是么

wan和lan没有互换的，互换很简单嘛几秒钟的事，没必要。

Wyaohui, 2020-12-07:
现在是usb网卡怎么办，发现切换N模式5G就不用，自动变成client。只能用legacy模式，而且信道不能自动，必须手动给，才可以启动

slb116763814, 2020-12-21:
@Wyaohui 都不支持5.8G频段，只支持2.4G频段的

zhzh8015, 2021-02-26:
遇到一样的问题，一样是2020-12-02Lean固件，一样是comfast 811ac网卡

5g模式设置为
模式：Legacy
频宽：5G
信道：149

2.4g模式设置为
模式：N
频宽：2.4G
信道：7
强制40Mhz频宽 开启
高级设置下MU-MIMO 开启

但是很不稳定，一开始能识别到网卡，并且为master模式，但是过一会儿网卡就显示：无线未开启或未关联

看到别的网友说用netgear a6100稳定些，有待验证

klever1988, 2021-02-27:
[image]

稳了 哈哈

zhzh8015, 2021-02-27:
> [image]
> 
> 稳了 哈哈

感谢老大回复，我现在用的固件是R2S-Lean-2020-12-02

请问是改用最新的R2S-Minimal-2020-12-14就能稳定使用comfast 811ac吗

klever1988, 2021-02-27:
> > [image]
> > 稳了 哈哈
> 
> 感谢老大回复，我现在用的固件是R2S-Lean-2020-12-02
> 
> 请问是改用最新的R2S-Minimal-2020-12-14就能稳定使用comfast 811ac吗

再等等 我在更新中